SignalK/specification #318: Outside light levels (Lux)

toqduj, 2017-01-08:
I'm making another MKR1000-based wireless SignalK sensor for mounting on deck, this one containing a humidity, pressure, temperature and light level sensor. I noticed there's no entry in the SignalK specifications for outside light levels. Such a reading would be useful, for example for automatically dimming instrument lights based on the available levels.

I'd suggest the following:
/vessels/<RegExp>/environment/outside/illuminance

Units: Lux (Lux)

Description: Current outside ambient light flux.

tkurki, 2017-01-08:
Makes sense to me. Would you be able to add this to the schema file and add a test?

toqduj, 2017-01-09:
I would, but I have no idea how (so would need to learn first), and too
swamped with work tasks to add this to my list. I'll have to take a rain
check on this one!

B.

rob42, 2017-02-18:
I will do it tomorrow.

rob42, 2017-02-23:
done and merged

closed 2017-02-23 by rob42