SignalK/specification #323: Source and derived values

bkp7, 2017-01-24:
Currently $source is not a required key, at least not for the most part.

Imagine an environment where there are multiple SignalK servers, perhaps some very simple slaves with just 1 or 2 sensors attached; multiplexers, displays and a Master somewhere in there, potentially each from a different manufacturer. Trying to make sense of data in this environment, particularly where some values will inevitably be duplicated will be very difficult if the source is not present. I would like to propose that the source be required. After all every device generating SignalK must surely be capable of telling us what it is.

Following on from this there are several values that might be calculated or derived from others eg. True Wind, Alarm/Notifications, etc. It is quite possible that the underlying data comes from one or more sources separate the the server doing the calculation. I propose that the source in this instance be the server which calculated the value. This will make it clear where the value has come from and the relevant server can decide what functionality it wants to expose to allow the user to choose which data the calculations are based on. I'm thinking here this would be in a multiplexer/master.

It means that if the server that produces the underlying data also happens to do the calculation that can be accommodated.

In the case of notifications a small server with a couple of sensors might have default zones which it can populate and generate notifications from. However if there is another server upstream which wants to allow these to be overridden by its own zones, the notifications from the original source can easily be identified and ignored.

Thoughts?

tkurki, 2017-01-31:
$source is not required to make it possible for a sensor to send it and the receiving server to fill in. I'd be totally ok in making it required for a server's output, but it can't be both - either use case needs to be spelled out in docs. Making it required would allow a bit stricter validation, which is probably more relevat for validating a server's output than for a sensor's.

SK data federation is still largely unexplored territory. I agree with your thinking.

bkp7, 2017-02-01:
I don't quite understand your distinction between a sensor and a server. We've been working on the basis that even the very simplest Signal K device must push out schema compliant json it must therefore be a (slave) server? I'm using the word sensor to mean the actual device, thermistor, or whatever, which is connected to a processor of some sort, which is configured as a server.

I'm working on a developer FAQ doc, because I think there are a number of areas where it could be clarified how things are meant to be implemented (source is one example). Shall I create a PR or do you want to make a branch for it. I'm certain it will need wide input.

rob42, 2017-02-18:
I dont think it needs to be mandatory. The current handling is that the server, or at least the next smartest signal K device adds it when its missing

.On the simplest devices the $ source is likely to be hardcoded anyway, so it will be of marginal value.

If its missing the receiver also needs to make a decision about trusting it.

There are a number of use cases where $ source has little value like self data onboard the same vessel, (we trust it anyway and only have one gps)
or maybe streaming race data to spectators, saving some bandwidth

when data travels off-vessel. lt becomes much more useful. Knowing who made the route you are following over a bar helps asses risk

bkp7, 2017-02-19:
I don't understand why $source isn't useful on board, and maybe I'm missing something here, because I though source need some more info in it than it has. 

At the display I would have expected that it should be possible to fairly easily use the source info to get to the setup pages on the source server. If there are a number of sensor servers on board and I want to change the zone/warning settings for a particular gauge, I would have hoped that a link to the server setup page could be available from the display device. If that is not the case, the user would be required to keep very detailed records of which sensors were on which servers. No very user friendly. So I was hoping that a link to the server http setup pages could be available in sources.

sumps, 2017-02-19:
$sources is very important onboard the vessel, it is amazing how many GPS sources can be on even a small boat !

I think it should be mandated for all sensors that output SK data to transmit $source even if the data is sent over serial/TCP/UDP or some other method other than WS Deltas.

There is an overhead involved in being a SK sensor/device and if they can support SK JSON then they should also be capable of outputting a $source but I think we need to define the format like we have for NMEA0183 and N2K devices.

rob42, 2017-02-19:
I probably didnt say it well, I meant that the data can be used without knowing source, it still works you just dont know as much about it. Thats not a killer on a small boat where you trust the few instruments you have. If you have a very simple sensor and you want it to have a meaningful source, then you also need to be able to configure it etc, which adds a lot of overhead. So easiest is to ping out a signalk string with a key and a var (temperature maybe). The next device (eg a server) can add timestamp and source, since it knows more about the transport path and context of the sensor (serial.tty1).
If we make `source` mandatory thats not possible, so we encourage a bunch of simple sensors with custom protocols.
I agree its an important attribute, just that it should not be mandatory. I would like to make it mandatory for servers and smarter devices, but that would make for awkwardly complex rules. Better that the implementations check for source, and decide for themselves if it matters in this case.

bkp7, 2017-02-24:
@rob42 if a sensor is connected to a server as part of one manufacturers system, the communication between the sensor and server can be any format. The server then puts out valid signalK. So the server can reasonably be expected to populate the $source.

If we imagine the simplest possible SignalK 'sensor', which sends out a SignalK JSON string for other upstream SiganlK servers to use, it will have to have a way of configuring the branch into which data must go, so it's not unreasonable to also expect it to be able to set a $source. I think that is what @sumps is saying.

I also agree with @sumps that the source format should be defined for SignalK devices. Any suggestions?

rob42, 2017-02-24:
My view is a bit different. In the case of the simplest sensor it could use a custom protocol, but then it only works with the same brand server. 

So the simplest sensor is better to send out any valid signalk delta, with a possibly unique source id , its the server that needs to recognise it and 'correct' the data. Then we can add the 'correctIncomingSensorData' capability to any server, and it works for all sensors. Even if the sensor has a non-unique source, it will still have a unique mac, serial or i2c address. So the server can differentiate it.

Anyway the point is that I dont really want to make `source` mandatory, because we dont _need_ to. Its better to send out `source`, sometimes even necessary, but not always.

BTW Ive added a minimal `source` format for `ais` in #337. We should keep adding as we learn

inspirity, 2017-02-24:
How about calibration of the source data? If the server has anyway to correctIncomingSensorData it would be great to have the possibility to have an offset and a factor value for a sensor source if needed.

rob42, 2017-02-25:
The 'correctIncomingSensorData' functionality would be a good way for applying offsets etc to really basic sensors data. Another alternative is to store and send a configuration to a more intelligent sensor, and have it do the changes. The key is that when the transfers are done in signalk, any sensor will work with any server, not just the same brand one.