SignalK/specification #329: Grouped Values in Schema lack sufficient detail.

bkp7, 2017-01-30:
Some values in the schema are grouped together and share a timestamp rather then the typical composite values. Examples are Current (see #41), Keel, Attitude, and possibly others.

The approach has been that since these values are 'linked' and will be updated together they don't each need a composite value.

However I think the logic is flawed. To group like this it must therefore follow that each element in the group can only be valued at the same time as the others (ie from one sensor), and that the group values can only be displayed as a whole.

Since meta data is now part of extended value (which I don't think was the case when this decision was made), there is no way to define what an instrument or associated warnings would be for each.

Taking each of the above in turn, current drift and set are often in 2 separate displays. For attitude, unlike in aircraft, roll and pitch are often in 2 separate displays, similarly it is hard to imagine a keel angle display presented as 1 instrument.

Final point, if this approach is valid for current, why is it not also valid for other linked data such as wind?

tkurki, 2017-01-31:
>can only be displayed as a whole

This part I did not get. The schema does not dictate display. A UI is free to pick how it displays stuff - for example InstrumentPanel has a wind gauge that shows wind speed & direction, that may arrive in different deltas and possibly from different sources.

My take on the logic here is that current is derived and not measured and both direction & speed are part of the derived value, whereas wind speed & direction come from two different sensors. @timmathews ?

What we are missing for composite values is metadata, for example zones. That is a real problem.

bkp7, 2017-02-01:
Agreed, schema doesn't dictate display. What I meant by 'can only be displayed as a whole' is that the ONLY way to display the composite data is within one gauge. So whilst a way to display current would be as a vector, the 2 elements could be displayed separately. Indeed for a river environment the directional element has very limited value so a display which just shows speed would be all that's required. Thus if there can be separate gauges, there needs to be separate meta for each.

As regards sensors, what about a sensor suite on a race buoy or wreck/diving marker? To measure current drift and set, that would need 2 sensors.

rob42, 2017-02-18:
the UI problem (needing individual meta ) could possibly be solved by a composite meta, eg individual meta values in the master meta
A bit clumsy maybe

bkp7, 2017-02-24:
Any further thoughts? It still seems to me that having a separate meta section for each would be the best solution, giving zones functionality, and dealing with it in the same way as other vector type data (ie separated speed and direction). @rob42, @tkurki, if you agree I am happy to implement.