SignalK/specification #333: Server-based units (and metadata in general)

tkurki, 2017-02-12:
We have currently some metadata elements as part of the `meta` element in schema ([docs](http://signalk.org/specification/master/data_model_metadata.html), [schema](https://github.com/SignalK/specification/blob/1117a7b54fd554fc95511eed8f49b89e1f7ebee5/schemas/definitions.json#L470-L597)) such as a few names and alarm/display related stuff.

However the schema files are enriched with units and explanation for each path. This data is distilled as [keyswithmetadata.json](https://github.com/SignalK/specification/blob/506fdf3bd7ae91479afcb148a054c63aacea3b2b/keyswithmetadata.json) and at least instrumentpanel uses that. I believe OpenPlotter plotter has some need for it as well.

So we have essentially two totally different ways to access similar data. Furthermore there is a fundamental problem with the schema-file based metadata: it is not available at all via HTTP. This means that clients can not query the server for metadata for data it provides, which rules out upgrading schema with new paths and their metadata. Instrumentpanel currently bundles the schema files / keyswithmetadata.json, which is not really an acceptable solution for Signal K in terms of extensibility.

I propose we come up with an extension of the current API that allows a client to query the static metadata for paths. I don't think this static data should be included always in the full http response along with actual values - I believe we need to start catering to different uses in the http API, one way won't fit all use cases.

tkurki, 2017-02-12:
Related to at least #244 and #206.

tkurki, 2017-02-12:
See also #130.

tkurki, 2017-02-12:
@sbender9 how do you handle units in WilhelmSK?

sbender9, 2017-02-12:
I bundle the whole schema in the application. Which I hate to do. Getting this from the server would be great.

tkurki, 2017-02-12:
@ph1l any thoughts to share?

bkp7, 2017-02-13:
The meta element in schema already has a units element which returns the SI unit for a value. It is not clear from the documentation whether this is the standard SI unit for the value being passed (ie what you are looking for) or the SI unit for display purposes. You're saying it's the former. However, since the meta element is about what is done with data (display/alarms) it would make sense that the display units be present in meta. Would it not make sense to have in the meta element both the SI unit and the display unit with a conversion factor?

In terms of #130, how about working with proforma meta elements at server level. So for example a fuel tank gauge can be set out within a proforma meta element in the server and reused for each fuel tank on board. Fuel quantity is a good example, as trying to second guess all the possible display units would be very difficult: US Gals, Imp Gals, ltr, lbs, tons, etc, etc. This way when a server and its sensors/sources are set up, a consistent set of displays, zones, warnings, and units is defined for that sensor/source. This approach I think extends the functionality proposed by #130.

Following on from this, I think the meta element is one of the most powerful features of SignalK. It means that repeater displays, whether portable or fixed, all share the same zones, gauge descriptions, language etc. ie. you set it up once when the sensor/server is installed and they will appear consistently on any device. Therefore making the meta element required, and also the SI unit required within it, would guarantee to give the SI unit to the display thus solving your issue? It would also probably encourage implementations of SK servers and displays/gauges to use the meta data fully.

sbender9, 2017-02-13:
I keep going back and forth on having the units preferences on the server side. It would be nice because then we can do things like put values in notifications in the users preferred units. But when something is using something like my app, you really want to display in the users preferred units. (If you're out with me on your boat, you may still want to see wind speed in m/s)

But there's also no reason we can't do both.

What's definitely missing for me in the spec is the ability query the server to find out what the units are for the data that I'm getting. Currently, the user needs to pick the units for a given gauge.

bkp7, 2017-02-13:
Absolutely agree that both can be accommodated. meta => vessel preferences, display software=> user preferences. 

Also agree you need to be able to get the  SI units from the server. My proposal is that sending the SI unit becomes mandatory for the server in the meta element, thus giving you the SI unit you need, along with zones, etc.

I have no particular issue with a separate API call to get the SI units, however it seems more sensible to me to include it in meta data, so when you set up the gauge's zones you also get the SI unit and preferred display unit.

tkurki, 2017-02-13:
@rob42 does java server support units and the other properties in meta?

rob42, 2017-02-13:
Only if you send them in. Like the node server it doesnt really care what the key is, it just stores it.
So you could populate the meta for keys when first seen,but would need a lookup table. Or you could send them from the lookup table when meta is requested,

ph1l, 2017-02-16:
My gut feeling is that the server should just keep SI units and it should be up to the client application to convert if needed/wanted. I have not used or looked at the notifications yet. But if an alarm is triggered, does it not have some way to refer back to the actual data item? wouldn't it be up to the client to deal with building a message in some language and some unit?

I keep thinking of the eventuality here. signalk servers on a larger network where multiple boats are sharing data over a wan. to have my boat in California saying it's 4 feet deep and your boat visiting the bay, but from Europe, saying the water is 1.1 meters seems confusing..

the data spec should be a standard that resists changes of convinience to maintain compatability and interoperability in the future.

ph1l, 2017-02-16:
To be clear:

The way the schema is defined externally seems to work okay. You just need to take into account that clients need to understand that part of the schema.

Some possibly off-topic thoughts:
 * Does signalk have a hard schema version that is stable and won't change?
 * Does the server publish that version reliably?
 * What is the process for breaking changes and backwards compatability in the server?

I see there is a tag for schema version 0.0.1-0 from Dec 2016. Is that a protocol version?

sbender9, 2017-02-16:
We should probably add the value as an attribute on the notifications, that would allow the client to display the value properly.

bkp7, 2017-02-17:
@ph1l the server always sends the data in SI units, I don't think anyone is suggesting otherwise. Delta's for example would always have values in SI units. This makes good sense and provides consistency when there is a wider network , as you point out. As I understand it the issue here is about a client understanding what those SI units actually are for any given element. So it's great saying the client can do conversion, but it needs to know what units it was sent. Currently it has to have a copy of the schema for that which it can't currently request from the server. Also even having the schema would not work for values not explicitly included in the schema.

sbender9, 2017-02-17:
Another issue here that we should take into account is that it's very hard for me to take a path like "electrical.batteries.0.voltage" and find that in the schema. (Unless I'm missing something)

sbender9, 2017-02-17:
Oh, forget that. I think as long as it's in meta, I won't have to worry about it.

rob42, 2017-02-18:
I think I understand now. Even though we all Know the value is an SI unit and needs conversion to the users preference, there is no way for Instrument panel to know that dC. voltage is obviously a volt measurement and therefore the SI unit is volt,
Easy for a human, and easy to hard code,but doing it dynamically requires a lookup

So we store it in meta which works. Getting it into meta for previously unknown keys either requires a lookup table, or the user needs to choose and save back to meta.
A lookup table is easy enough but there will always be exceptions, so a panel to allow use selection of SI unit is still required.

rob42, 2017-02-18:
As far as getting a copy of the schema or a lookup  I dont think the client should,
Just request the key meta, up to the server implementation to lookup the unit, if it cant then pop the user selection box and let them choose

tkurki, 2017-10-14:
Fixed in 5aa38cf3e6e80caf2ada7657f33dc42445090523 and https://github.com/SignalK/signalk-server-node/pull/294

closed 2017-10-14 by tkurki