SignalK/specification #334: Add Bearing and Distance Fields to Navigation Group

sumps, 2017-02-18:
If iKommunicate is receiving GPS and AIS data, it currently adds a Distance field and Bearing field to the Navigation Group of any AIS targets it receives. We use these fields in our iKompass App to show where AIS targets are and they could be used in TCPA and CPA alarms. Here is an iKommunicate Delta message showing these fields...

`{
    "context": "vessels.urn:mrn:imo:mmsi:368610000",
    "updates": [{
        "source": {
            "type": "NMEA2000",
            "src": "001",
            "pgn": 129038,
            "label": "N2000-01.001"
        },
        "timestamp": "2016-07-02T17:38:39Z",
        "values": [{
            "path": "navigation.position",
            "value": {
                "longitude": -122.391129,
                "latitude": 47.666508,
                "distance": 15009.578125,
                "bearing": 4.267966
            }
        }, {
            "path": "navigation.speedOverGround",
            "value": 0.000000
        }, {
            "path": "navigation.courseOverGround",
            "value": 3.804803
        }]
    }]
}`

I would like to add these two fields to the Navigation group as optional values for other vessels received via AIS. Comments ?

sumps, 2017-02-18:
Does anyone know how to paste formatted JSON from jsbeautify in to GitHub without losing the nice formatting ?

sumps, 2017-02-18:
Makes sense to add CPA and TCPA in the Navigation Group as well. These could be populated by a server along with notifications and then it removes some quite complex calculations from the web app.

rob42, 2017-02-18:
to pretty print json wrap it in 3 backticks. Hard on a cell phone my keyboard doesnt have one😡

rob42, 2017-02-18:
No problem with ikommunicate using them but I don't see them as usetul in the model

The problem is the values are only useful in relation to the vessel with the AIS receiver, If they are then sent on to another vessel they would be wrong, possibly dangerously.

sumps, 2017-02-18:
If it is clearly defined, then I don't see it as anymore dangerous than any other data transmitted off the boat, it is just another measurement specific to the boat that sends the data. As an example the depth measured by one boat is specific to that boat and by the same argument could be dangerous if another boat received it.

I think it has some value as a derived measurement, as would CPA and TCPA and I would prefer that it was defined in the spec, rather than hidden away as iKommunicate's secret.

rob42, 2017-02-19:
Looking at it again you are right, the values received would be in the 'other' vessels navigation group, so you would see that _they_ were  xNM at xxxdeg from target y.
I was stupidly thinking that the data would end up in vessels.self on the receiving vessel, fooling the crew to thinking target y was at xNM at xxxdeg from themselves.

rob42, 2017-02-19:
Can you add a pull and test?

sumps, 2017-02-19:
Yes I will try and do that this week.

tkurki, 2017-02-19:
For one vessel the meaning is clear, they are relative to 'self', but if these end up in an aggregating server the self is less clear.

How about navigation.relativePosition.selfId with two properties: bearingTrue, distance. This would allow multiple relative positions and makes it clear what it is relative to. Except which is the origin is not quite so obvious.

The same logic would allow cpa and tcpa under navigation.approaches or something similar.

rob42, 2017-02-19:
It will break in aggregation without both vessel ids.

The bearing and distance will be relative to the ais sending vessel (which we know) and the ais receiving vessel, which may be in $source, or may be overwritten by some intermediate server. 
The concept of bearing and distance is only valid when it includes the two vessels, and then for consistency we could (should) be sending a list of vessels and bearing/distance to all neighbours, gets pretty ugly quickly. What sort of data would we get with 100 AIS targets also sharing signalk?

I think I am back to my first position - that its an implementation specific value that might be derived by a given server for its own efficiency. I dont think its viable to propagate beyond the originating vessel.

sumps, 2017-02-20:
I think it is better to keep the data clearly and simply stored in the schema with no convoluted key or extra properties and if people want to aggregate it in the cloud then they will write code and find methods to do so. Let us not over complicate the schema for niche applications we cannot even think of yet, if the data is clearly stored in the schema people that want to use it will do the work involved to get the data out in the way they want it.

rob42, 2017-02-20:
The bearing and distance between any two vessels can be easily calculated using their lat/lon ( vincenty calc, https://en.wikipedia.org/wiki/Vincenty's_formulae). But if we try and pass that data between vessels it must include both vessels or it is inherently confusing and error prone.

The only way to keep it 'clear and simple' is to define the key as _always_ meaning "distance and bearing _from self_". Then a server implementation can strip/check the value and correct it if it comes from an off-vessel source.
I think thats probably the best solution if we want to have these keys in the spec.

rob42, 2017-03-11:
Im closing this  as its bogging down in complexity. The values _can_ be calculated easily for those that need them,  this is really just an implementation optimization. Not needed for V1 - lets push on :-)

closed 2017-03-11 by rob42