SignalK/specification #352: Keys list generation does not work

tkurki, 2017-04-30:
Generating http://signalk.org/specification/master/keys/ no longer works because of the very large number of keys added when the sar/aton/aircraft introduced so much duplication
- technically it fails with stack overflow when doing the .md => .html transformation
- usabilitywise it also fails: the list is goddamn long and has a lot of wasteful duplication

tkurki, 2017-04-30:
BTW this prevents the whole documentation publishing without taking action to remove the keys list or do something about it.

rob42, 2017-04-30:
Maybe the keys can be split to multiple docs, one each for sources, resources, vessels, aircraft, atons, etc?
That would reduce duplication per page?

toqduj, 2017-05-01:
Or by taking a clue from programming: defining a base class / key list, and the application-specific derivatives thereof for each of sar/aton/aircraft.

tkurki, 2017-05-01:
@toqduj looking forward to schema structure improvement PRs. There is already considerable reuse, but JsonSchema is not the most amenable tool for that. It also makes the schemas less human readable.

toqduj, 2017-05-01:
But we can at least separate the key lists for the documentation, no? One general key list, and the application-specific modifications / additions to it?

tkurki, 2017-05-01:
Yep, sounds like a plan. Now who's up for implementing that?

sailoog, 2017-05-02:
I guess this is also affecting keyswithmetadata.json file. We are doing an intensive use of this file in openplotter to create menus and assign signal k keys to the multiple connected sensors.  Please do not discard this file. We are also  missing enum values in this file, they were present in the past.

tkurki, 2017-05-04:
@sailoog no, this is about generating the html docs.  Please add a separate issue for keyswithmetadata.json.

PR #357 (2017-06-07): Split gitbook doc keys list
A very crude way to move forward with the docs: split the single
keys list to vessels and others docs.  Fixes #352.

Ping @rob42 - after this #355 can move forward.
closed 2017-06-07 by tkurki