SignalK/specification #356: Navtex

joabakk, 2017-05-31:
Just a thought, should Navtex messages have a place in the specification?

sbender9, 2017-06-01:
I could google it. But might make sense to have the info here. What are Navtech messages?

joabakk, 2017-06-01:
Sure:
Navtex (Navigational Telex) is an international automated medium frequency direct-printing service for delivery of navigational and meteorological warnings and forecasts, as well as urgent maritime safety information to ships. It can look like this: 
`Navtex Bulletin Received (UTC):
2017-06-01 00:01:05

ZCZC AA84
310820 UTC MAY 17
COASTAL WARNING NOVOROSSIYSK 283
BLACK SEA
1. GUNNERY FIRING EXERCISES 02 THRU 04 JUN FROM 0500 TO 1800 UTC
NAVIGATION TEMPORARILY DANGEROUS IN AREA BOUNDED BY 43-27.0N
039-46.7E 43-30.0N 039-41.6E 43-31.4N 039-42.8E 43-26.7N 039-50.7E
2. CANCEL THIS MESSAGE 041900 UTC JUN
NNNN

A003
www.navtex.net`

toqduj, 2017-09-04:
These are sent out regularly, and may be received using the same SDRs that we use for AIS data. I do not know of any suitable demodulation plugins, however. Navtex is an essential part of the GDMSS and SOLAS regulations, and so is a good candidate for inclusion in SK. 

You would also want to have flags for which message types / subjects you are interested in. Not all messages are of interest to all sailors. 

More info: https://en.wikipedia.org/wiki/Navtex

joabakk, 2017-09-24:
The regular SDR will not wok, as they are outside the frequency range. But using a device such as this https://swling.com/blog/2016/04/mikes-overview-and-review-of-the-40-soft66rtl3-sdr-by-kazunori-miura-ja7tdo/ and a method like this https://groups.io/g/linuxham/topic/4851931?p=,,,20,0,0,0::,,,0,0,0,4851931 might work. But hard and software solutions aside, I think it could be a valuable addition

toqduj, 2017-09-30:
It would be a worthy addition, as I don't think there's any talk of phasing it out just yet. 

I've been playing around with the SDRplay RSP2, but haven't managed to catch any navtex signals, let alone decode them.

mgrouch, 2021-04-16:
There are decoders. Free and open source ones. Like this one:

https://arachnoid.com/JNX/index.html

There are also Inmarsat Fleet messages 

And decoder:

https://bitbucket.org/scytalec/scytalec/src/develop/

I’ve built in them into BBN Marine OS
https://github.com/bareboat-necessities/lysmarine_gen

mgrouch, 2022-02-08:
Inmarsat-C decoder is here

https://github.com/cropinghigh/stdcdec

mgrouch, 2022-02-08:
There is software to read NasaMarine NavTex receiver via serial usb

https://github.com/juerec/pc-navtex