SignalK/specification #359: Polar diagram representation

joabakk, 2017-06-04:
In working with polars, a sort of table is required. For a specific wind speed, a table is made for the specific vessel. https://github.com/joabakk/signalk-polar does this by a database method, but has no place in the specification to store the table itself, only some of the main values.
If https://github.com/SignalK/freeboard-sk/issues/13 is to succeed or any other implementation of the same, such a table should be available in the schema in order to standardize.
Ideally, this should be open for any relative wind angles, to cater for any source of this table without interpolating or need to discard good data. I'm not sure this contradicts https://github.com/SignalK/specification/blob/cb728f95f8912b11b066ee8d05ffa329004bbf61/gitbook-docs/design_notes.md, and how this best would be captured in an organized manner, otherwise I would have done a PR

rob42, 2017-06-04:
Ah found my old code :-)
I stored the data by recording a point every time I crossed a 15deg heading boundary and stored it in a structure like:
`"performance.polar.[segment].wind": vesselSpeed` where [segment] is 0, 15, 30, 45... degrees. Wind was rounded up, but value was recorded for a segment. When a better result was obtained for the same segment/wind it was replaced.
 
That was enough for Kevin Kemps algo to derive the rest of the polar. The more you sail, the more accurate the polar.

rob42, 2017-06-04:
Hmm....given the Signalk-IMU gives attitude data, you could derive seastate, and have different polars for different wind/sea conditions.... @raffmont - a thesis for one of you students?

joabakk, 2017-06-05:
That is one approach, but given that the available data will vary in resolution, and that starting to use degrees here where the rest of the spec (sauf position) is in radians, it may violate the predictability of the paths. Here is an extract of an ORC certificate, where the main angles are most usually not in that 15 deg interval. I am storing any better data with the same wind speed and true angle, which makes the resolution close to infinity. There are many ways to skin a cat, but how do we best represent such a table in the schema?
[image]

rob42, 2017-06-05:
deg vs radians - yes, but the degrees are just a marker and the equiv radians are not an int. It was easy at the time :-)
You can use the same technique down to 1 deg or even smaller, or any random sample of angles, but there was no real value in it for my case. With the kemp algorithm I used it only requires a dozen random points at a few different wind speeds to derive the full polar curve, and hence all the other values for the table.

rob42, 2017-06-05:
eg once you have recorded a suitable collection of points you can query the algorithm for the target speed for any other windspped/angle.

rob42, 2017-06-05:
Reading your question again - the table does not have to be stored in the schema - only the data to allow it to be generated on demand. Hence using my data and the Kemp algorithm outputting the table is straight-forward.

tkurki, 2017-06-06:
Once again you are confusing API definition with persistence methods: it does not really matter if the data is coming from a database, extrapolated with a particular algorithm or pulled out of thin air. 

We are defining _what data and in what format_ would a client get polar data from a server's http/rest interface.

I think there is a legitimate case for making polar data from a particular source available in the SK data model as is, or as close to as is as possible, and not force a particular algorithm on everybody.

For this case I would be tempted to create a structure that includes some metadata and data:
- name, description
- angle units: deg | rad
- polar data

and the full data model must support having multiple polar sets.

@joabakk how is your JSON Schema? I think this can be tackled with first considering the polar data & metadata separately, and when happy with it we can hang it off the full model where appropriate. So create a schema/groups/polar.json separately and write a few test cases / examples for it.

joabakk, 2017-06-06:
Thanks for the thoughts. It is getting clearer to me as we think aloud. So for the current combination of:

- true wind angle relative to the boat `environment.wind.angleApparent`
- wind speed `environment.wind.speedApparent`

There exists a `performance.polarSpeed`. What we need is a way to poll the equivalent polarSpeed of another combination of angle and  wind speed and perhaps sails set, heave motions etc. That way, it would be implementation agnostic and it would be up to the implementation to choose static tables with inperpolation and all the way to fully dynamic/algorithm based lookup.
The basic example wit an API responding to a query combination of wind angle and speed should be straightforward (I think). This would correspond to the assumption that if you are not able to reach your polar speed, you need to change sails to achieve it (ie combined polar). Or you could devise a reduction factor for heave motions and  add that to the schema.
Once the server can respond to such queries, one could poll the necessary value combinations to create a table on the fly.
It could be as simple as `performance.correspondingPolarSpeed.[windSpeed].[angleApparent]`, but that would not support multiple polar sets (as in sail combination, sea state etc).

@tkurki I am happy to suggest, but I feel this discussion hasn't fully converged yet, so I'd like to hear more from both of you or others first?

rob42, 2017-06-06:
@tkurki - Im not suggesting the Kemp algorithm is forced on every-one. Just making the point that if you store data in a structure like I used, then other values can be extrapolated from there. Since there are a a huge number of `[angle].[wind]`, combined with seastate and sails, its not really feasible to collect and store values for them all and then do a simple lookup. You have to use some sort of maths to fill in the blanks anyway.

I dont think the Signalk API should define any specific formatted output. For me it needs to answer two basic questions:
1) What is my target speed for this `seastate.angle.wind` combination?
2) Get full polar data for `seastate`. Really a quick way to do 1) many times, so we can export/graph it or whatever

Both can be answered with our current API if we store the data as discussed.

In the end a polar diagram is a radial graph, and we only need to store as many points as required to connect the dots and give the rest. To allow for many polars we could use the same structure as elsewhere
`performance.polar.[name].angle.wind` where name is either well-known or arbitrary, or we could do `performance.polar.[seastate].angle.wind`, or both `performance.polar.[name].[seastate].angle.wind`

joabakk, 2017-06-07:
I like your thinking @rob42 . I think `performance.polar.[name].[seastate].angle.wind` is perhaps the cleanest and most expandable one. 

- name can be anything from a named combination of sails to year etc, or "all" to skip this step
- Seastate should be something simple for lookup, such as Beaufort (integer 0-12), perhaps -1 to show all?
- Angle should be integer, ie degrees or even tenths of a degree
- wind speed is the normal m/s

rob42, 2017-06-07:
[name] could be a boat design name eg Benetau31. Then we could upload polar to a central store and share 😊

raffmont, 2017-06-07:
@rob42 : about sea state from IMU: it's one of our topic of interest (we have another one on it: predict engine faults from a 3 axis accelerometer). Let me check if there is a student available.

joabakk, 2017-06-07:
Here is an attempt to structure the schema side of this. I've moved angle to the end, I think it makes more logical sense
```{
	"polar": {
		"type": "object",
		"description": "An object describing the polar diagram of a sailboat.",
		"title": "polar",
		"properties": {
			"name": {
				"type": "string",
				"description": "An unique identifier by which the polar diagram can be recognized",
				"example": "Maxi_Fenix_fullMain_fullGenoa"
			},
			"seastate": {
				"type": "string",
				"enum": [
					"0", "1", "2", "3", "4", "5", "6", "7", "8", "9", "10", "11", "12", "all"
				],
				"description": "The Beaufort sea state, see https://en.wikipedia.org/wiki/Beaufort_scale",
				"example": "all"
			},
			"WindSpeedTrue": {
				"description": "Wind speed over water (as calculated from speedApparent and vessel's speed through water)",
				"$ref": "../definitions.json#/definitions/numberValue",
				"units": "m/s"
			},
			"angleTrueGround": {
				"description": "True wind angle based on speed over ground, negative to port",
				"$ref": "../definitions.json#/definitions/numberValue",
				"units": "rad"
			}
		}
	}
}
```

tkurki, 2017-06-07:
If we start with real code I suggest opening a PR with [WIP] in the name. Then we can use Github's tools to advantage, like per line commenting.

I did not realise that sea state would be Beaufort scale - isn't that just another unit for wind speed?

In your schema `name`  is really not a name, but an id. And the fact that you are trying to encode information into it means that there should be a human oriented `description` and in addition specifically the set of sails and their reef state should be encoded properly. How's our sail inventory and current sails configuration, can we reuse stuff from there?

Your stab at schema is lacking the actual value.

`angleTrueGround `- Isn't the reference here water, not ground? Polars should look totally different relative to ground if there is a current.

tkurki, 2017-06-07:
While having the server interpolate for any wind speed & angle there is a definite use cases for retrieving the true underlying, usually tabular data:
- transport: send to friend via email, retrieve from a cloud server and retrieve it
- having some magical algorithm do it client side on the raw data
- i want to draw polar graphs - how do you I know what values are available? try all possible combinations? Polars are not defined for all combinations

See for example data in https://commons.wikimedia.org/wiki/File:Downwind_polar_diagram_to_determine_velocity_made_good_at_various_wind_speeds.jpg

I think the full set of values should be retrievable and a server may support querying for any combination. How should the server report back that "otherwise your http request is ok, but I have no polar data (extrapolation) for this combination" as opposed to "I have no such polar you're referring to"?

As per Joachim's schema above we could define the polar schema so that it can be used independently of the full SK stuff.

What if the structure looked somethiing like this? 3.5 and 4 are windspeeds, 30/35/40 etc wind angles.

```
{
    "xStd": {
        "name": "Manufacturer",
        "description": "Polars from X Yachts 2013, standard main & gyb"
        "data": {
            "angleUnit": "degrees",
            "polars": {
                "3.5": [
                    [
                        30,
                        2.75
                    ],
                    [
                        35,
                        3.1
                    ],
                    [
                        40,
                        3.2
                    ],
                    [
                        45,
                        3.25
                    ],
                    [
                        50,
                        3.43
                    ]
                ]
            }
        }
    },
    "my2016": {
        "name": "Own 2016",
        "description": "Derived from 2016 sailing data with Joachim's plugin"
        "data": {
            "angleUnit": "degrees",
            "polars": {
                "3.5": [
                    [
                        30,
                        2.55
                    ],
                    ...
                ],
                "4": [
                    [
                        30,
                        2.67
                    ],
                    ...
                ]
            }
        }
    }
}
```

joabakk, 2017-06-07:
I will PR a WIP, just have to get time to do it. My intention was to provide the structure for the API query, and the response should follow that structure. Beaufort scale is used to represent sea state (ie waves) in this context. We could use significant wave height in meters, but as Beaufort is integer and easier to interpret, that is what I used. But it does get messy to mix degrees and Beaufort and other non-SI units. Hm, I'll be back

rob42, 2017-06-07:
I dont think SI units is a big issue here as these are not measuremnts, but identifiers. It happens that meters and radians are floats at the required resolutions, which is awkward.

As you say Beaufort is an integer, and well documented. All we are really after is partitioning sea state from calm to horrible. Same for degrees. Its not really about the number, but the segment of the circle. We could use 1-12 (eg hours) or something, it just happens that deg is a handy int that has the required resolution.

tkurki, 2017-09-08:
https://github.com/sbender9/signalk-derived-data/pull/7

closed 2021-05-15 by joabakk