SignalK/specification #373: Unit ratio 0-100 or 0-1

e-sailing, 2017-08-01:
In most parts of the documentation it says:
propulsion.*.engineLoad                Engine load ratio, 0<=ratio<=1,    1 is 100%
So ratio should be a value between 0 and 1.

tanks.*.*.currentLevel                     Level of fluid in tank 0-100%
environment.inside.humidity          Current inside air relative humidity
These sentence have also the unit ratio!
They are used in n2k-signalk pgn 127505 and 130311. But they have a value of 0-100.
It isn't good to have ratio in two dimensions!!
Should we correct n2k-signalk to 0-1?

tkurki, 2017-08-01:
Yes we should. Can you help out with that?

e-sailing, 2017-08-02:
I did a pull request on n2k-signalk.
130311 was OK only the json file was not ratio.
127489 was OK.
Only 127505 and the test were not ratio.

tkurki, 2017-08-04:
Is there more to do here or should we just continue at https://github.com/SignalK/n2k-signalk/pull/82 ?

e-sailing, 2017-08-06:
This is solved.
Thank you for the help. I learnd much.

closed 2017-08-06 by e-sailing