SignalK/specification #398: True Wind vs. Ground Wind

ronzeiller, 2017-11-26:
In the specifications I found:

## environment.wind.angleTrueGround
True wind angle based on speed over ground, negative to port

"environment.wind.speedOverGround": {
"shortName": "TWS",
"longName": "Wind Speed Over Ground in Meters/Second"

If you take **TWS as Ground Wind** what is taken for **True Wind Water referred** which is taken for VPP (polar data) calculations?

tkurki, 2017-11-26:
http://signalk.org/specification/0.0.1-12/doc/vesselsBranch.html#vesselsregexpenvironmentwindangletruewater

ronzeiller, 2017-11-26:
Ok, thank You, this specification doc I was not looking at

So just to clarify:
**/vessels/<RegExp>/environment/wind/directionTrue
/vessels/<RegExp>/environment/wind/directionMagnetic**
is probably meant the Ground Wind?

In https://github.com/SignalK/specification/blob/master/src/i18n/en.json
I found (and this is, why I was asking):
 **TWS ** in conjunction with **AWS** is more commonly used as the water referred True Wind Speed and Apparent Wind Speed which is also shown at instruments.
```
"environment.wind": {
    "shortName": "Wind",
    "longName": "Wind Data"
  },
  "environment.wind.angleApparent": {
    "shortName": "Angle Apparent",
    "longName": "Apparent Wind Angle"
  },
  "environment.wind.angleTrue": {
    "shortName": "Angle True",
    "longName": "True Wind Angle"
  },
  "environment.wind.directionChangeAlarm": {
    "shortName": "Direction Change Alarm",
    "longName": "Wind Direction Change to Raise Alarm"
  },
  "environment.wind.directionTrue": {
    "shortName": "Direction True",
    "longName": "True Wind Direction"
  },
  "environment.wind.directionMagnetic": {
    "shortName": "Direction Magnetic",
    "longName": "Wind Direction Magnetic"
  },
  "environment.wind.speedAlarm": {
    "shortName": "Speed Alarm",
    "longName": "Wind Speed Alarm in Meters/Second"
  },
  "environment.wind.speedTrue": {
    "shortName": "Speed True",
    "longName": "True Wind Speed in Meters/Second"
  },
  "environment.wind.speedOverGround": {
    "shortName": "TWS",
    "longName": "Wind Speed Over Ground in Meters/Second"
  },
  "environment.wind.speedApparent": {
    "shortName": "AWS",
    "longName": "Apparent Wind Speed in Meters/Second"
  },

```

tkurki, 2017-11-28:
The i18n files are an earlier start towards providing documentation and localization. It has been abandoned and I’ll remove it shortly. Replacement is documentation in schema files and the annotated key list in the documentation that I linked and keyswithmetadata.json.

closed 2017-11-28 by ronzeiller