SignalK/specification #399: specification changes

dmdelorme, 2017-11-28:
I am trying to understand how to add, append, change. Keys ie  electrical.json and keyswithmetadata.json. Does one create the other or do i have change both files and what other ones do I have to modify.

tkurki, 2017-11-28:
The master is in schemas, but unless you understand JSON schema well you don’t start with the schema files but open an issue and discuss the proposed additions. Then somebody can carry out the changes and add test/demo cases.

dmdelorme, 2017-11-29:
The changes i have made are local with the understanding I will lose them with any update. 
I am getting better with JSON. 
so far i have issues with 
 electrical.solar.*.loadVoltage  I could not find this in the schema? 
electrical.batteries.*temperature does not have a units.
Would prefer electrical.solar.*.batteryTempature.
As the unit has several temperature sensors with their own warning and fault values. but only 1 batterytemp sensor.
I have a fairly long list of values that the charge controller holds to shoe horn into SignalK. Some will never be read like Carbon dioxide reduction L/h..

bkp7, 2018-02-28:
The solar section in the electrical schema has a number of values in common with charger:
`electrical.solar.*.voltage` which is the output voltage from the charge controller
`electrical.solar.*.current` the output current from the charge controller
`electrical.solar.*.setpointVoltage` which is the target regulation voltage
`electrical.solar.*.chargingMode` which can be bulk/float/etc

In addition solar has specific solar properties:
`electrical.solar.*.panelVoltage` being the voltage from the Solar Panel array
`electrical.solar.*.panelCurrent`
`electrical.solar.*.panelTemperature`

Is the loadVoltage you refer to different to the output voltage? ie. can the load port be at a different voltage to the charger circuitry? If so I would agree that there should be a loadVoltage which I am happy to write the json for.

---
`electrical.batteries.temperature` does have units (K). This is defined in dcQualities.
eg.
```json
{
  "uuid": "urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d",
  "electrical": {
    "batteries": {
      "main": {
        "temperature": {
          "value": 291.2,
          "timestamp": "2018-02-28T09:46:55.302Z",
          "$source": "foo.bar"
        },
        "chemistry": "FLA"
      }
    }
  }
}
```
---
If you can supply a list of real world properties, I'll make an example from them.

dmdelorme, 2018-03-01:
It would be nice if we include power too.
Although this is a derived value  and could be calculated many charge controllers output this.
electrical.solar.*.panelPower
electrical.solar.*.power  .... output of charger in Watts
electrical.*.power  ..... consumption "load" in Watts

bkp7, 2018-03-01:
The Signal K model happily accepts derived values in many other areas, so I can see no reason not to allow a derived value here. A device which only outputs power (without voltage and current) should be able to include that in Signal K.

I will put together a full example and do a PR for the changes to the schema. Do you have any other values you want to incorporate. You mentioned "a fairly long list ... to shoe horn into SignalK" which if you can provide we can discuss here.

dmdelorme, 2018-03-02:
I get Volts and Amps and display them it is just cleaner to look at watts.
The epsolar outputs lots of data but what is really relevant is another question? I dont need to know how many tons of co2 i have saved.. 
it has 3 sets of inputs/ outputs 
power out of panel, charge into battery and Load. plus Battery temperature.
I am currently 1500km from the unit but will be home on Tuesday.
the unit also stores daily, monthly and yearly totals in watts used and produced which is handy. Well the yearly is useless on a boat but the daily/ weekly is handy for consumption budgeting.
When i get back home i will re evaluate what is useful and what is just fluff.  Ie as i just built a masthead unit that sends wind data and controls the Anchor light via code and low power wireless connection i could in theory turn the anchor light off when the panel voltage reaches X. I have a switch wired to a gpio port on the pi that sends a code to the arduino when it is on/off and sends a message to signalK that the anchor light is on or off.

dmdelorme, 2018-03-07:
This is my output to signalK.
`"{"updates":[{"source":{"type":"solar","src":"epsolar","label":"OPserial"},"values":[{"path":"electrical.solar.Tracer4210A.voltage","value":12.66},{"path":"electrical.solar.Tracer4210A.current","value":0},{"path":"electrical.solar.Tracer4210A.power","value":0},{"path":"electrical.solar.Tracer4210A.panelVoltage","value":0.34},{"path":"electrical.solar.Tracer4210A.panelCurrent","value":0},{"path":"electrical.solar.Tracer4210A.panelpower","value":0},{"path":"electrical.solar.Tracer4210A.loadVoltage","value":12.66},{"path":"electrical.solar.Tracer4210A.loadCurrent","value":0.34},{"path":"electrical.solar.Tracer4210A.loadPower","value":4.3},{"path":"electrical.chargers.Tracer4210A.loadnet.current","value":-0.34},{"path":"electrical.chargers.Tracer4210A.mode","value":0},{"path":"electrical.batteries.Tracer4210A.temperature","value":293.15},{"path":"electrical.batteries.Tracer4210A.capacity.stateOfCharge","value":0.56}],"timestamp":"2018-03-07T10:58:48.528Z"}],"context":"vessels.urn:mrn:imo:mm..."
3/7/2018, 4:58:48 AMnode: 257d5f6.e0e5d2msg.payload : string[216]
"{"updates":[{"$source":"OPsensors.1W.inside1w","values":[{"path":"environment.inside.refrigerator.temperature","value":294.962}],"timestamp":"2018-03-07T10:58:48.649Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765432"}"
3/7/2018, 4:58:48 AMnode: 257d5f6.e0e5d2
msg.payload : string[1004]
"{"updates":[{"source":{"type":"solar","src":"epsolar","label":"OPserial"},"values":[{"path":"electrical.solar.Tracer1210A.voltage","value":12.63},{"path":"electrical.solar.Tracer1210A.current","value":0},{"path":"electrical.solar.Tracer1210A.power","value":0},{"path":"electrical.solar.Tracer1210A.panelVoltage","value":0.19},{"path":"electrical.solar.Tracer1210A.panelCurrent","value":0},{"path":"electrical.solar.Tracer1210A.panelpower","value":0},{"path":"electrical.solar.Tracer1210A.loadVoltage","value":12.63},{"path":"electrical.solar.Tracer1210A.loadCurrent","value":0},{"path":"electrical.solar.Tracer1210A.loadPower","value":0},{"path":"electrical.chargers.Tracer1210A.loadnet.current","value":0},{"path":"electrical.chargers.Tracer1210A.mode","value":0},{"path":"electrical.batteries.Tracer1210A.temperature","value":298.15},{"path":"electrical.batteries.Tracer1210A.capacity.stateOfCharge","value":0.56}],"timestamp":"2018-03-07T10:58:48.974Z"}],"context":"vessels.urn:mrn:imo:mmsi:987654..."`

 The code:
`import socket
  sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
  SignalK = '{"updates": [{"source": {"type": "solar","src" : "epsolar"},"values":['
  SignalK += '{"path": "electrical.solar.'+str(unitid)+'.voltage","value":'+str(round(batVolts, 2))+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.current","value":'+str(batchgAmps)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.power","value":'+str(batchgpower)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.panelVoltage","value":'+str(round(pvVolts,2))+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.panelCurrent","value":'+str(pvAmps)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.panelpower","value":'+str(pvpower)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.loadVoltage","value":'+str(ldVolts)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.loadCurrent","value":'+str(ldAmps)+'}'
  SignalK += ',{"path": "electrical.solar.'+str(unitid)+'.loadPower","value":'+str(ldpower)+'}'
  SignalK += ',{"path": "electrical.chargers.'+str(unitid)+'.loadnet.current","value":'+str(ldnetAmps)+'}'
  SignalK += ',{"path": "electrical.chargers.'+str(unitid)+'.mode","value":'+str(chargestate)+'}'
  SignalK += ',{"path": "electrical.batteries.'+str(unitid)+'.temperature","value":'+str(batTemp)+'}'
  SignalK += ',{"path": "electrical.batteries.'+str(unitid)+'.capacity.stateOfCharge","value":'+str(soc)+'}'
  SignalK += ']}]}\n'
  sock.sendto(SignalK, ('localhost', 55559))`

This  is output from a cli script currently not sent to signalK.

> /dev/ttyXRUSB0
Tracer4210A
power consumed:  110.0
power generated:  400.0
running: True
chargerfault: False
charge stage: Off  : 0
Input volt status: Normal  : 0
charging mode:  mppt
battype Flooded
batcapacity 150
battempcomp 3
highvoltdisconnect 16.0
chargelimitvolt 15.0
overvoltreconnect 15.0
equalizevolt 14.6
boostvolt 14.4
floatvolt 13.8
boostreconnectvolt 13.2
18 / 3 / 6 18 : 29 : 15.0
pvpower 0.0
batpower 0.0
ldpower 5.79