SignalK/specification #401: Magnetic Variation

sumps, 2017-11-29:
In the spec we do not specify what the variation is E or W. Anyone with NMEA experience and knowledge could assume but I think it is best to state this clearly to avoid mistakes.

Positive Variation should be Easterly and Westerly Variation negative so that when you add the variation to the Magnetic Heading you correctly get the True heading.

tkurki, 2017-11-29:
Could you come up with exact wording to update  "description": "The magnetic variation (declination) at the current position?

sumps, 2017-11-30:
I would suggest...

"The magnetic variation (declination) at the vessel's current position that must be added to the magnetic heading to derive the true heading. Easterly variations are positive and Westerly variations are negative (in Radians)."

bkp7, 2018-03-04:
This issue can now be closed.

closed 2018-03-04 by tkurki