SignalK/specification #405: Metadata problems

bkp7, 2018-01-06:
Quote from Specification: "This metadata may include information such as the unit of measure, minimum and maximum permissible values, alarm thresholds and localized display name for every data point in the model.". The Schema says: "Provides meta data to enable alarm and display configuaration". However, it does not seem to achieve these objectives.

Problems with Metadata:

1. The overall gauge limits are not given anywhere.

Possible solution:
Add a lower and upper gauge limit to the meta properties. I considered whether we could use a whole of gauge zone, but since that would be a zone with unique properties (ie. both upper and lower limits required) I have discounted it. Also there is no way that the gauge 'zone' could be considered an alarm zone. One benefit of specifically defining gauge limits as opposed to trying to infer gauge limits from warning zone limits, is that alarm conditions can be independent of the gauge. For example a typical coolant temperature gauge might display from 40 to 120 Celcius with warning zones at the higher temps. It might be useful to have a warning zone if for example the coolant temperature were below freezing in order to warn against engine start, even though that warning zone would not be depicted on the gauge.

2. There is no explanation as to what should happen if a value is in 2 zones simultaneously. Which should take precedence? We cannot rely on the order they appear in the json array. Also there is no explanation or requirement that zones are contiguous. What is the position where the value is not within any zone? The examples given in the documentation all have overlapping zones because the upper and lower limits are by schema definition within the zone.

Possible solution:
Either force every zone to have both an upper and lower limit and make sure they are contiguous, or explain that zones should be considered by their level of severity with higher severity overriding lower ones. My preference is the second option as it requires no changes to the schema or usage. Also it would seem sensible to explain that anything not within a zone is considered normal, therefore any normal zones supplied would have no effect.

3. The are 4 free text descriptions: displayName, longName, shortName, description. There is no explanation as to what/where these should be used.

Possible solution: As a minimum an explanation and use case for each should be documented within the description in the schema. Possibly they are not all needed. eg. 'displayName': 'the text which appears on the face of a gauge to describe what it is measuring.', 'description': 'a more detailed explanation of the gauge/value in question and might for example be accessed by a user hovering over the gauge', etc. Any suggestions as to what the possible uses of longName and shortName are? 

4. There is a suggested "gaugeType" but no indication as to possible/permissible values. The only 2 examples I can find in the docs are analog or sparkline. Is it not expected that other gaugeTypes be implemented? It is described in the schema as "the gauge necessary to display this data", but the documentation describes it as the "recommended display format".

Solution: Make the documentation consistent with the schema description. Also consider whether it should be an enum, or there should be a suggested list. Without some guidance the gaugeType will only work where server and client are from the same manufacturer.

5. The units property is explicitly defined to give the SI units for the value. There is nowhere to supply the recommended or shipwide units for the value. Without this every gauge on every display must be set up individually for the preferred display units. This seems counter to the objectives of SignalK and since SI units are rarely the preferred display units it will be very onerrous on users.

Possible Solution: add a property of displayUnits which can be non SI units. Similar to displayName and gaugeType, this is the suggested unit of display. In doing this there would be a need to standardise on possible non SI units to prevent variants, eg. "Kn", "Knot", "Knots". This could be implemented as an array of display non SI units with the definition of each SI unit.

7. Timeout is described in the schema as "The timeout after which this data is invalid", but the documentation says it can also be used as the interval between polling for a new value. These seem to be 2 distinctly different requirements being rolled into one value. What value is to be used for invalid data if the gauge is polling. It must be greater than the polling interval to allow time for a response to its poll? As an example a fuel gauge might poll once a second, but due to the nature of the measurement that could be valid for many seconds or even minutes before the gauge has to cease showing any value.

Possible solution: ensure the documentation is consistent with the schema. Add a new property of pollingInterval and some explanation against both. 

8. Some gauge styles have 'null zones', typically near the centre of the normal band within which the indicator doesn't move at all. It enables gauges to be designed so that all needles even across different measurements to be aligned during normal operations making identification of any trends away from normal very easy to spot. This is not currently supported in SignalK at all.

Possible Solution: Add an additional zone type of 'dead' or 'null' which can be used to indicate such a zone if relevant to the gauge design in question. An alternative would be to use 'normal', which assuming the proposed solution to 2 above were adopted, would otherwise have no meaning (because anything which was not specifically a warning is now considered 'normal').

If I get some feedback on the above I will get the changes implemented.

tkurki, 2018-01-07:
Wow, lots of good feedback!

Protocolwise I see one problem here: while all of these concern metadata they are not really related in the sense that they can or should be fixed at the same time. If one of these subissues does not get resolved within a reasonable time frame this Github issue will live long and not prospoer, because discussion about the different things will get intermixed and drag on and on. Could you please create separate issues / continue discussion as outlined below?

I consider all of these separate issues, possibly linked but resolvable separately:

>The overall gauge limits are not given anywhere.
>if a value is in 2 zones simultaneously
>The are 4 free text descriptions: displayName, longName, shortName, description. There is no explanation as to what/where these should be used.
>There is a suggested "gaugeType" but no indication as to possible/permissible values. 
>Timeout is described in the schema as "The timeout after which this data is invalid", but the documentation says it can also be used as the interval between polling for a new value. 
>Some gauge styles have 'null zones'

and then

>There is nowhere to supply the recommended or shipwide units for the value. 

A very good point that needs resolving, but this is not quite so simple, as you might prefer depth in fathoms, boat lenght in feet and distance in nautical miles, but the SI unit is meters for them all. See https://github.com/SignalK/specification/issues/130, continue discussion there?

bkp7, 2018-01-07:
Issues separated as you suggested. Closing this one now.

closed 2018-01-07 by bkp7