SignalK/specification #406: Overall gauge limits are not included in metadata

bkp7, 2018-01-07:
Whilst zones are defined in the meta data the upper and lower limits of the gauge itself are not defined. Existing implementations have to make assumptions about the gauge limits eg. some implementations assume 10 knots as the maximum for STW which is not useful for faster vessels.

Possible solution:
Add a lower and upper gauge limit to the meta properties. I considered whether we could use a whole of gauge zone, but since that would be a zone with unique properties (ie. both upper and lower limits required) I have discounted it. Also there is no way that the gauge 'zone' could be considered an alarm zone. One benefit of specifically defining gauge limits as opposed to trying to infer gauge limits from warning zone limits, is that alarm conditions can be independent of the gauge. For example a typical coolant temperature gauge might display from 40 to 120 Celcius with warning zones at the higher temps. It might be useful to have a warning zone if for example the coolant temperature were below freezing in order to warn against engine start, even though that warning zone would not be depicted on the gauge.

tkurki, 2018-01-14:
Yes, how about adding something lilke
```
"displayScale": {
  "lower": 10
  "upper": 15
}
```

bkp7, 2018-01-14:
Or:
```
"gaugeLower": 10,
"gaugeUpper": 15
```
or:
```
"gaugeScale": {
  "lower": 10,
  "upper": 15
}
```
At first I didn't like yours because 'scale' meant to me some sort of sizing. However on reflection I think it may be better not least because it could also include whether or not the scale on the gauge should be logarithmic. eg. depth, luminosity (I know it's a separate point and I've not raised an issue for it yet).

I'm not keen on display though. To me one display or instrument panel can have multiple gauges, see similar usage in subscription_protocol.md (line 90-112).

If we can decide on the terminology I will implement it.

tkurki, 2018-02-05:
I think we have used the word display elsewhere in the schema. To me it is more related to displaying the value”, more a verb than a noun.

bkp7, 2018-02-06:
You're quite right, 'displayName' is used already in metadata in the context of one gauge and could be interpreted as either a verb or noun.

In that context 'displayScale' sits alongside nicely, so unless these are any other thoughts I'll put it in as you suggested.

In passing 'display' is also used in the context of units definitions (eg "display": "Hz") which I don't think bears on this at all.

closed 2018-05-03 by bkp7