SignalK/specification #407: metadata: gauge zone overlap

bkp7, 2018-01-07:
There is no explanation as to what should happen if a value is in 2 zones simultaneously. Which should take precedence? We cannot rely on the order they appear in the json array. Also there is no explanation or requirement that zones are contiguous. What is the position where the value is not within any zone? The examples given in the documentation all have overlapping zones because the upper and lower limits are by schema definition within the zone.

Possible solution:
Either force every zone to have both an upper and lower limit and make sure they are contiguous, or explain that zones should be considered by their level of severity with higher severity overriding lower ones. My preference is the second option as it requires no changes to the schema or usage. Also it would seem sensible to explain that anything not within a zone is considered normal, therefore any normal zones supplied would have no effect.

tkurki, 2018-01-14:
See the example here https://github.com/SignalK/instrumentpanel/pull/21 - in this case zones with just one bound are useful.

As for overlapping zones your idea of using severity order sounds good to me. Here we are talking about notifications/alarms, right?

Why would the zones have to be contiguous? Again see the example I linked, and there's the case that no zones are even defined.

In Signal K data model zones are connected to nofitications. Other than that they have a semantic meaning and a Signal K client may choose to treat them how they want.

bkp7, 2018-01-14:
Yes, that's a good example and I agree zones with just one bound are useful.

Regarding overlapping zones/severity it would apply to notifications/alarms and also to the colouring/zoning on a gauge, because one value can only be one colour. ie I would expect that in the example below the area from 14.2 upwards would be shown as red (alarm) rather than green (alert) even though 14.2 - 14.5 are declared in both zones:
```
{
  "zones": [{
    "lower": 12.8,
    "upper": 14.5,
    "state": "alert",
    "message": "Charging"
  }, {
    "lower": 14.2,
    "upper": 15,
    "state": "alarm",
    "message": "Voltage too high"
  }]
}
```
I also don't think zones should be contiguous, but the presence of a "normal" state in the schema could lead one to think they should be. Taking the example you gave, there is no zone labelled as normal and indeed including one (or more) would have no effect on either the gauge rendering or any warnings.

I think that 'normal' (ie. getting no notifications) should be removed as a state.

PS #411 regarding null zones is not relevant to this discussion, see my comments over there.

PR #463 (2018-06-07): feature: add 'nominal' state to meta zones
Add `nominal` state to zones within meta and update documentation accordingly.

closes #411 and resolves #407
closed 2018-06-07 by fabdrol