SignalK/specification #409: metadata: gaugeType not well defined

bkp7, 2018-01-07:
There is a suggested "gaugeType" but no indication as to possible/permissible values. The only 2 examples I can find in the docs are analog or sparkline. Is it not expected that other gaugeTypes be implemented? It is described in the schema as "the gauge necessary to display this data", but the documentation describes it as the "recommended display format".

Solution: Make the documentation consistent with the schema description. Also consider whether it should be an enum, or there should be a suggested list. Without some guidance the gaugeType will only work where server and client are from the same manufacturer.

tkurki, 2018-01-14:
Since gaugeType has no meaning for the server the problem is more about if it is desirable or even actually feasible to have a global gaugetype per data item. What if you want to have one gauge analog and another digital? Wind as close hauled or full range gauge?

All in all I don't think this is a very useful field and would consider removing it.

bkp7, 2018-01-14:
I think you're right and agree with removing it. I think it would be better implemented as some sort of default or per gauge style of instrument on the client end and left to the designer of the client to deal with as they wish.

bkp7, 2018-05-03:
Giving this further thought I have 3 scenarios where it would be useful to have this. If we think about consumers they will range fro the very simple, to the sophisticated. Looking at Wilhelm for example there are many different gauges which are specific to particular values eg. a rose for compass, vertical and horizontal gauges for pitch and roll, etc, etc. So in this case the consumer has an internal map of branches to gauge designs.

1. The sophisticated consumer encounters a branch for which it has no internal mapping, what should it use as a default?

2. The sophisticated consumer has a choice of gauges for any one particular branch. It would be useful if the choice was coordinated between consumer instances.

3. A simple consumer has no map at all of branches to gauge type

In considering this we should bear in mind the stated rationale for meta:
```
In an environment where various critical pieces of information are displayed in multiple locations it becomes quite
difficult to ensure that all of these devices use the same scale and react the same way to changes in the data. This is
especially true in an environment where these devices are not tied to the boat. A crew member may bring a personal
tablet with them for their tactician role during a Wednesday evening race or a harbor pilot may bring a laptop on board
loaded with local charts. If these devices can load critical configuration data from a central server on the boat, this
saves time and prevents costly or even disastrous mistakes from occurring due to misconfigured devices.
```
Therefore I propose that `gaugeType` be retained with enumerated values of: analogue, digital, rose, sparkline and that wording be put together explaining that the consumer is not obliged to use the `gaugeType` value but is encouraged to do so if it would add consistency between consumers bearing in mind the design, look and feel of the consumer gauge suite.

bkp7, 2018-06-08:
@tkurki @fabdrol I have done a PR (#469) for this but appreciate no-one else has commented on my approach. Could you take a look please.

fabdrol, 2018-06-12:
@bkp7 I'm not sure that gaugeType solves the (potential) problem that comes from the stated rationale. IMO wether a value is displayed in an analogue, digital, rose or sparkline fashion doesn't matter: the (potential) problem to solve is that a critical value *is displayed in the correct units*, and how it's shown by a consumer really doesn't matter that much (e.g. I can read a value, and discern it's meaning, in any of the fashions mentioned by the `enum`). 
Furthermore, if a server would specify, let's say, an analogue gauge - set by the captain - and a crew member who performs a critical job (trimmer) prefers a digital gauge because of the added detail the `gaugeType` may even create the aforementioned problem (assuming the crew member doesn't change the settings in his consumer).

There's another issue to consider here: adding this to the specification just adds to the amount of bloat, and it has the potential of increasing the work a user has to put into server setup. It also raises a question in the development of consumers: if server defines a gaugeType, should we follow it or allow it to be overwritten? In the latter case, what's the point of the gaugeType in the first place (assuming one accepts my argumentation above). 

Personally, I feel that we should aim to keep the domains `server` and `consumer` pure (at least from a specification point of view) and try to prevent overlap as much as possible. If there is valid use case (I do see some merit in @bkp7's scenarios) to "sync" consumers of a similar type (or consumers on the boat in general), I feel that this should be handled as an implementation-level API, separate from specification.

bkp7, 2018-06-12:
@fabdrol, I think that the metadata concept of Signal K is one of its 'game changing' features. The rationale in the documentation is spot on:

"In an environment where various critical pieces of information are displayed in multiple locations it becomes quite difficult to ensure that all of these devices use the same scale and react the same way to changes in the data. This is especially true in an environment where these devices are not tied to the boat. A crew member may bring a personal tablet with them for their tactician role during a Wednesday evening race or a harbor pilot may bring a laptop on board loaded with local charts. If these devices can load critical configuration data from a central server on the boat, this saves time and prevents costly or even disastrous mistakes from occurring due to misconfigured devices."

This is very neat and powerful and allows devices to be set up consistently throughout the vessel, but also allows personal devices or individual stations to use personalised/customised langauge, units, layout, even style/suite of display.

I have come to the conclusion that there are 2 types of items in the meta data:
1. fixed data which should always be used by the display consumer ie. red line limits, warnings, etc
2. vessel wide display defaults, ie. language, units, etc.

At present we have ```displayName``` which is very much in the second category and allows the name of the item to be shown in a consistent vessel wide language/locale. I certainly think that putting the language dictionaries on the server is much better than having each consumer have to implement their own. Our own (WIP) server implementation for example has a global language selection with default displayName, longName and ShortName for each data point which goes in meta for the consumer to use.

We see the displayUnits as being the same thing, I would not imagine users setting each one up on the server individually, but they could if they wanted to change vessel wide units for particular data points. In terms of usability I think having the displayUnits #464 in meta is an extremely powerful/useful addition.

To answer your point about increasing server setup, I don't think this would be the case unless someone wanted a very bespoke vessel wide approach. I'd envisage server designs coming up with their own way of setting these defaults easily (eg. language/locale setting), or a simple server may just not bother.

For the consumer, if displayName, or displayUnits are present it should use that as the default but allow the user to set something else if they want. One could imagine a tactician creating a very bespoke and personalised display which completely ignored these defaults.

I do think gaugeType is far more tenuous than adding displayUnits. As it stands gaugeType is freetext and I cannot see any practical use for it in the current format. You will see that @tkurki and I suggested complete removal and I think there is merit for that approach. My proposal to make it an enum also I think has some merit given the use of the server for setting vessel wide defaults. Happy to go either way.

fabdrol, 2018-06-12:
@bkp7 thanks for your comprehensive approach. I don't disagree with your point about the power of a central place to store this kind of information, my main objection is that it seems really messy to me to have two sources of truth: the central server's metadata capabilities, and the local device persistent storage. 

Having said that, however, it's only my personal opinion and I'm not insensitive to other views and valid use cases, so let's go with the flow there. Regarding gauge types, however, I think getting rid of them is the right way to go - I feel that the `enum` approach will always be incomplete (as consumers come up with innovative new views, do we keep adding them to the spec?) and allowing any string value defeats the purpose. 

Could you change the PR to remove it, then I'll confirm the merge.

bkp7, 2018-06-12:
@fabdrol happy to go with your proposal.

I would just say that regarding adding items to the enum I did give that some thought as I agree consumers could come up with all sorts of display types. However the 4 chosen are deliberately very generic and I would not anticipate needing to add to them in future.

Having said that I will create a PR to remove gaugeType. This will require a version bump to 1.1.0.

fabdrol, 2018-06-12:
@bkp7 should it, in that case, not be in schema cleanup?

fabdrol, 2018-06-12:
Alright, will you open a new PR and should we close this one, or will you use this same PR so we retain the discussion?

bkp7, 2018-06-12:
I was thinking of changing this PR to note deprecation of gaugeType, and do another PR to remove it on the schema cleanup branch.

PR #470 (2018-06-13): chore: remove deprecated gaugeType
Remove gauge type from meta.

closes #409
PR #469 (2018-06-13): chore: mark gaugeType as deprecated, update documentation and samples
resolves #409
closed 2018-06-13 by fabdrol