SignalK/specification #410: metadata: timeout appears to have a dual purpose

bkp7, 2018-01-07:
Timeout is described in the schema as "The timeout after which this data is invalid", but the documentation says it can also be used as the interval between polling for a new value. These seem to be 2 distinctly different requirements being rolled into one value. What value is to be used for invalid data if the gauge is polling. It must be greater than the polling interval to allow time for a response to its poll? As an example a fuel gauge might poll once a second, but due to the nature of the measurement that could be valid for many seconds or even minutes before the gauge has to cease showing any value.

Possible solution: ensure the documentation is consistent with the schema. Add a new property of pollingInterval and some explanation against both.

bkp7, 2018-04-25:
The section is [here](https://github.com/SignalK/specification/blob/c375cd7e4c50200a1380eb688c6409bec2e05d1a/gitbook-docs/data_model_metadata.md#metadata-for-a-data-value)

I agree with your comments on polling. Maybe we just need a re-wording here. What you describe could well have been what the author was getting at, but I didn't read it that way.

How about:

'The `timeout` property tells the consumer how long it should display a value for before it must be refreshed. If a new value is not received the consumer must assume the source has failed and display an appropriate error indication. If the consumer is using polling to read the value the `timeout` may be used as a basis for deciding on a polling interval. This value is specified in seconds .......'

tkurki, 2018-04-26:
You are dictating client behavior and thinking just about displaying the latest value.

 _The timeout property tells the consumer how long it should consider the value valid_ would be more accurate.

It is really up to the consumer (and possibly the human user who configures the consumer) how it chooses to deal with the situation: flash the display, just clear the display, start drawing a dashed instead of a solid line in the graph, stop drawing the line, change the vessel marker on the map, remove the value from the latest values cache....

tkurki, 2018-04-26:
...and I think we should just remove the reference to polling. That is up to the client.

PR #460 (2018-04-27): docs: clarify timeout definition/usage
Change documentation for `timeout` within metadata to remove reference to polling and bring in line with description and usage as described within the schema [definitions.json](https://github.com/SignalK/specification/blob/e87d62c7856ddb723048389cf8e335a564821cd2/schemas/definitions.json#L577).

closes #410
closed 2018-04-27 by tkurki