SignalK/specification #411: metadata: no provision for gauge null zones

bkp7, 2018-01-07:
Some gauge styles have 'null zones', typically near the centre of the normal band within which the indicator doesn't move at all. It enables gauges to be designed so that all needles even across different measurements be aligned during normal operations making identification of any trends away from normal very easy to spot. This is not currently supported in SignalK at all.

Possible Solution: Add an additional zone type of 'dead' or 'null' which can be used to indicate such a zone if relevant to the gauge design in question. An alternative would be to use 'normal', which assuming the proposed solution to issue #407 were adopted, would otherwise have no meaning (because anything which was not specifically a warning is now considered 'normal'). The only issue then would be backward compatibility.

tkurki, 2018-01-14:
"normal" is in alarmStates already. Is this a spec issue, or just up to somebody to implement gauges that work the way you describe, using a zone with state `normal` to specify the null zone?

bkp7, 2018-01-14:
What I am getting at here is differentiating 2 types of 'normal'. I'm defining 'normal' to mean no warnings or alarms (as does SK). In this null zone the gauge doesn't move even when the value does. A common use of this is for engine monitoring eg. coolant temperature where there is a normal (no warnings) (green) zone between say 70C and 110C, but when the temperature is between 80C and 90C the needle doesn't move at all (typically remains vertical or horizontal). This is really useful if you have many gauges (multiple motors with multiple sensors) where it is very easy to spot that every needle is pointing in exactly the same direction, or not!....

Reflecting on comments on the other meta issues, I think it would be unwise to re-use 'normal' state for this purpose and therefore I would favour a new (optional - as all are) state called 'null'. I think null zone is a good description of what it does and is similar to established use in joysticks where moving the joystick results in no movement of the controls.

tkurki, 2018-04-22:
I think you've made the case for adding this, but I have to say that I have a serious dislike for "null". Null often means the absence of a value, which this is not. `null` in JSON is also very different from `"null"` and we use `null` as a value denoting that the source is working, but can not produce a value.

So something else, please. I agree that this is a different "normal". I am not a native english speaker, but how about `"nominal"`? Merriam-Webster's 5th meaning for that is _being according to plan : satisfactory; "everything was nominal during the launch"_.

bkp7, 2018-04-25:
Yes, agreed, nominal is much better. I will produce a PR with appropriate changes.

PR #463 (2018-06-07): feature: add 'nominal' state to meta zones
Add `nominal` state to zones within meta and update documentation accordingly.

closes #411 and resolves #407
closed 2018-06-07 by fabdrol