SignalK/specification #423: is value property required?

bkp7, 2018-01-18:
Where there are multiple sources of the same data we can have values eg:  (valid against vessel schema)
```json
{ "uuid": "urn:mrn:signalk:uuid:c0d79334-4e25-4245-8892-54e8ccc8021d",
  "propulsion": {
    "instance0": {
      "revolutions": {
	"timestamp": "2014-08-15T19:00:15.402Z",
        "$source": "source.a",
        "value": 1280,
        "values": {
	  "source.a": {
	    "value": 1280,
	    "timestamp": "2014-08-15T19:00:15.402Z"
	  },
	  "source.b": {
	    "value": 1281,
	    "timestamp": "2014-08-15T19:00:15.731Z"
}}}}}}
```
The schema currently allows neither, either, or both: value and values. Is this correct?

Should it be valid without either?
Should it be valid with only values?

NB. values are set up consistently in the schema, so the above is equally true for all value types.

Some guidance or a reference to understand whether or not this is correct would be appreciated.

tkurki, 2018-01-19:
Value is requires. It is the oocation of the _default_ value for that data item. Default is up to server policy, but anyway a safe path for a client to retrieve a value, as there must always be something there.

Values signals the existence of multiple values and is required if there are multiple values. Its existence is not defined for the single value case.

bkp7, 2018-01-19:
Great, that's what I would have expected. See PR #425

bkp7, 2018-06-13:
The requirement to have a value has been implemented in PR #425 at least for items utilising commonValueFields, as part of the schema cleanup project.

The issue regarding whether values should be optional for all data points is transferred to issue #427.

Therefore I am closing this issue.

closed 2018-06-13 by bkp7