SignalK/specification #427: Should Values be available in all cases?

bkp7, 2018-01-20:
Following on from #423, there are a number of cases where there is a "value", but there is no capability to allow "values" in order to cater for multiple sources. This occurs frequently in the schema where a value has not been defined in terms of the global definitions eg numberValue, but where the value is itself more complex (object) and therefore it seems that the values part just hasn't been implemented. Some examples are:

atonType
aisShipType
environment.mode [day, night, restricted visibility]
course.nextPoint
course.PreviousPoint
(lots more)

Logically if a branch in the tree can have a value, it can be provided from more than one source and so should have the possibility of values too. Is this correct?

rob42, 2018-06-05:
I think you are right in saying that if a path has a  `value` then it can have `values`. Looking at atonType as an example it would seem logical that an aton can only be one type. But that doesnt mean that every-one agrees that the aton is the same type :-( so we could definitely have `values` 

I suggest that as this is a breaking change, (v1.1) we need to start a v1.1 branch and make it in there. 
The v1.1 branch can have the v1.0.x changes merged in until its worth putting out 1.1

@tkurki ?

bkp7, 2018-06-08:
Sorry, closed this by mistake.

This needs to be dealt with on the schema-cleanup branch. @fabdrol could you set up a project for the cleanup and assign this to it. Thanks.