SignalK/specification #430: Broken link in  specification/gitbook-docs/data_model.md

lburaggi, 2018-01-28:
The "Source" link in data_model.md page is broken (specification/blob/master/gitbook-docs/sources.md) the sources.md page doesn't exist

tkurki, 2018-01-29:
The link was added in 246306620f6754cf832f636d6d50865af2c8f21d. 

@timmathews did you ever write sources.md? I can't remember there being one. Probably should write one, as this is the third or fourth question about it.

Ping @hkapanen.

bkp7, 2018-02-13:
Also data_model.md has a section in it about sources and $sources indicating that you can use either one. This is not possible as $source is a mandatory requirement for all values (from common values in definitions.json)

timmathews, 2018-02-13:
Yeah, kinda dropped the ball on the sources section. Writing one up now, a draft should be available shortly the a sources branch.

BTW, what is the point of [these lines in sources.json](https://github.com/SignalK/specification/blob/3222f988680c42d451fca742863845ee373082f8/schemas/groups/sources.json#L82-L88)?

timmathews, 2018-04-26:
PR #432 addresses this

closed 2018-04-26 by timmathews