SignalK/specification #437: tests: there is no check in place for schema validity.

bkp7, 2018-02-20:
There are currently no automated tests in place for schema validity and these should be added to prevent recurrence of invalid schemas (issues #434 and #320).

The current json validator in use (TV4) does not check for schema validity as part of its tests.

rob42, 2018-05-03:
Needs http access to the different versions of the schema (v1.0.0 differs from v1.0.4 ), currently I only see v1.0.0.
I'm very short on time, so I dont want to start on this if there is another solution underway, should I go ahead?

webmasterkai, 2018-05-03:
https://ajv.js.org might be a useful tool in validating.

bkp7, 2018-05-03:
@rob42, sorry perhaps my post was slightly misleading. There is no problem in getting the existing TV4 validator to validate the schema, it just that such a test hasn't been implemented yet.

All other json validation tools are useless to us at present as none support the TV4 'banUnknownProperties' flag which our schemas require in order to work properly (see issue #246).

We already have an online validator tool at [ http://node-master.signalk.org/@signalk/playground/dist/ ]( http://node-master.signalk.org/@signalk/playground/dist/ ) which uses TV4.

The current proposal is to update the existing schema to be draft-04 compliant thus removing the dependency on TV4. At that point any json validator can be used for validation and the proposal is to move to AJV. This will also give us coverage stats for our tests.

rob42, 2018-05-03:
ah - ok. I'll leave it then