SignalK/specification #439: clarification: are 'units' and 'description' required properties in 'metadata'

bkp7, 2018-02-20:
The Documentation refers to [Implicit Metadata](http://signalk.org/specification/1.0.0/doc/data_model_metadata.html#implicit-metadata). Does this mean that these 2 properties are always required within metadata? If not always, under what circumstances are they required?

timmathews, 2018-02-20:
Yes, `units` and `description` are always required and the schema should reflect that. What I was trying to convey in that section of the documentation was that values for `units` and `description` would be pulled from the schema itself and that end users should not need to provide them when configuring their specific metadata (i.e. zones, min/max, custom labels, etc).

bkp7, 2018-02-20:
OK that makes sense for the all the numeric values which have units.

However some values are enums but still have metadata eg. propulsion.instance0.state. In this case I'm not convinced it should have a meta section at all? What would the units be?

Other values are strings and also have metadata eg. tanks.fuel.main.extinguishant and again what would be inserted for units?

timmathews, 2018-02-20:
You're right it doesn't make a lot of sense for string and enum typed values. The docs could probably use a bit of clarification there. Docs should also be expanded to indicate that the possible enum values should be included in `meta` for enum typed values.

PR #462 (2018-05-13): fix: modify schema to enforce `description` and `units`
Schema modified to enforce the written specification that description and units must be present in meta.

Also added possibility for meta to include an enum list of possible values where the value is non numeric. This is not mandatory as at 1.0.x but the documentation notes that it is likely to become mandatory in future (see issue #439).

Samples and tests corrected to comply with the requirement.

Additional tests added to test the schema modifications.

Although the schema is not backwards compatible (making `description` mandatory whe [... cut]
closed 2018-05-13 by tkurki