SignalK/specification #457: GeoJSON issues

gilbertbw, 2018-04-11:
This line of the geojson spec seems to reference a file that no longer exist: https://github.com/SignalK/specification/blob/3228ead31730f997ea087dd1139d7703f94c24bf/schemas/external/geojson/geojson.json#L13

The json-schema example for geojson seems to be [here](http://json-schema.org/example/geo.json) but there is a discussion on the json-schema project about whether or not json-schema.org should be hosting reference schemas:
json-schema-org/json-schema-org.github.io#171 

Also the README for the geojson directory in SignalK
https://github.com/SignalK/specification/blob/3228ead31730f997ea087dd1139d7703f94c24bf/schemas/external/geojson/README.txt#L1
Shows that is is based on [this project](https://github.com/fge/sample-json-schemas) that appears to be un-maintained and in fact links to a [version of geojson that is obsolete](http://geojson.org/geojson-spec.html) and has been superseded by [RFC 7946](https://tools.ietf.org/html/rfc7946)

tkurki, 2018-05-06:
Now that I look into this a bit more: I don't see any references to external/geojson/geojson.json, it seems safe to remove it. 

```
schemas$ ag external
groups/resources.json
156:                      "$ref": "../external/geojson/geometry.json#/definitions/lineString"
272:                          "$ref": "../external/geojson/geometry.json#/definitions/polygon"
287:                            "$ref": "../external/geojson/geometry.json#/definitions/polygon"
```

rob42, 2018-05-06:
I originally added the copy of the current geoJson schema into ../external as I could not get the docson based signalk docs to work etc without it. If thats now fixed it can go

rob42, 2018-05-14:
docson is no longer used. 
Caching the spec would be a better solution.
I suggest changing it to reference the appropriate geojson url, then rerun tests and the various tools we use to publish docs etc. If it all works then lets merge it in.