SignalK/specification #474: REST requests for multiple values

sbender9, 2018-06-23:
I think we need a way to request a full tree via the REST api for a subset of values. 

This would be very useful for something like my Apple Watch app. Downloading the full tree on a boat like mine or making multiple request for many values is very expensive and slow. 

It does not seem like there's a standard way to do this via REST, so I'd appreciate any input. Otherwise I will come up with something and make a proposal...

tkurki, 2018-06-23:
Instead of cooking up something homegrown we should go with the standard solution: HTTP/2. Multiple requests with minimum overhead is one of its main use cases: HTTP/2 multiplexes multiple requests and responses over one persistent tcp stream, doing away with the major source of overhead, setting up and tearing down individual tcp connections.

With this approach we do not need to devise something new, just use the existing REST stuff.

https://github.com/spdy-http2/node-spdy

sbender9, 2018-07-11:
Agreed and PR submitted to add support for http/2 to node server.

closed 2018-07-11 by sbender9