SignalK/specification #481: No way to specify location of server web interface in endpoints

bkp7, 2018-06-27:
Currently the _http._tcp server web interface can only be defined in the optional DNS-SD advert, and if specified there it MUST be at the root.

There is nowhere to specify it in the endpoint message (discovery.json schema). Should there be an additional advertised service for a web interface? Should it be possible to locate a web interface elsewhere than in the root eg.  at `http://<<host>>:<<port>>/signalk/admin`?

tkurki, 2018-06-27:
I don’t understand what you are saying. A tcp service has no path concept, so _root_ has no meaning in this context. 

Tcp has nothing to do with http as used in this context - tcp refers to direct tcp socket connection, not http on top of tcp. 

Web interface and /signalk/admin is never mentioned in the specification afaik. And as a side note current node server interface is not even located at /signalk/admin.

A web interace is not part of the spec. The spec specifies the api.

What parts of the spec do you refer to?

tkurki, 2018-06-27:
Sorry, my mistake. _http._tcp specifies the web interface, if there is one.

Endpoint message is for rest interface - why should it be specified for anything other than the rest interface?

There is already a way to advertise a web interface. Please note this service name is not sk specific, just a way to advertise a generic web interface. I don’t think it needs to be anywhere but at root - if you need to support non root web app advertise a separate port and redirect from there to the actual port and path.

tkurki, 2018-06-28:
Yep, please clarify the doc, as it refers to the api endpoints.

Web interface ad makes sense to support browsers that have integrated Bonjour discovery. Safari used to have that, not sure if it still has.

If there is no Bonjour a user must have knowledge of the server’s address. In the most usual case the user would just enter the address in the browser. 

There is the case where you would enter the server api address in a non browser client which could provide a ’button’ to open the web ui based on data in endpoints. Not a huge thing to add it there.

bkp7, 2018-06-28:
> There is the case where you would enter the server api address in a non browser client which could provide a ’button’ to open the web ui based on data in endpoints. Not a huge thing to add it there.

@tkurki, agreed.