SignalK/specification #482: No way to permit secure interfaces other than by redirect

bkp7, 2018-06-27:
Currently the only permitted endpoints are:
- signalk-http
- signalk-ws
- signalk-tcp

Should these be permitted:
- signalk-https
- signalk-wss
- signalk-udp

Any others ?

Redirect doesn't seem like a great idea, it effectively means assuming all consumers support SSL/TLS

bkp7, 2018-07-12:
node-server implements https by pointing signalk-http at a secure endpoint effectively assuming that all consumers can support https. If it is the intention that all signalk consumers using the REST API MUST support secure connections I will add this to the documentation, but it strikes me that this is not a good solution effectively ruling out lightweight consumers.

An alternative would be to do this (permitted by current schema):
```
{
  "endpoints": {
    "v1": {
      "version": "1.0.4",
      "signalk-http": "http://192.168.1.5:80/signalk/v1/api/",
      "signalk-http": "https://192.168.1.5:443/signalk/v1/api/",
      ...
    }
  }
}
```
or this (not currently permitted by schema, but the point of this issue):
```
{
  "endpoints": {
    "v1": {
      "version": "1.0.4",
      "signalk-http": "http://192.168.1.5:80/signalk/v1/api/",
      "signalk-https": "https://192.168.1.5:443/signalk/v1/api/",
      ...
    }
  }
}
```
and make provision of an unsecured endpoint mandatory on servers implementing the REST API.

See also issue #485 relating to using duplicated keys in discovery.

sbender9, 2018-07-12:
I do not think that an unsecured endpoint should be mandatory for servers. Servers MUST have the option of only supporting https.

https support should be mandatory for clients.

I don't know of any json implementation that supports multiple keys with the same name and frankly should not be valid json in IMHO,

tkurki, 2018-07-12:
Not supporting both http and https was a practical choice, i did not consider spec implications. I think they should be configurable separately in the node server and available concurrently in the discovery doc if the system chooses to have both available. Redirect to https might fall outside spec scope.

I can fix this in the server but that will be >month away.

JSON documents with multiple properties with the same name are a Bad Idea. Let’s not go there.

Agree on https being mandatory for clients.

bkp7, 2018-07-23:
@tkurki, I have modified the docs and for the time being left it such that a server can support either http or https but not both ie the current (undocumented) position. I agree with you that a server should be able to concurrently run both http and https, but also it should be able to run 2 different versions of signalk (see #482). I think that to solve this properly will require a redesign of the discovery doc which effectively means waiting for v2. In the meantime no changes required to node-server, existing implementations or the schema.