SignalK/specification #496: Add a variable for 'maximum depth that the sounder can detect a sea-bed'

sarfata, 2018-07-27:
According to Actisense, the DPT sentence can include 'the 'maximum depth that the sounder can detect a sea-bed' (see page 13 of [this doc](https://www.actisense.com/wp-content/uploads/2017/07/NMEA-0183-Information-sheet-issue-4-1-1.pdf)).

I do not think there is a SignalK path to store this information at the moment. The signalk-nmea0183 parser does not parse this: https://github.com/SignalK/signalk-parser-nmea0183/blob/master/hooks/DPT.js#L27

Example (same source): 

> Standard precision, without checksum:
> `$SDDPT,76.1,0.0,100,<0D><0A>`
> 76.1 metres, zero depth offset, 100 metres maximum depth range.

Opening this issue to keep track of this but this is totally not urgent and I have yet to see an equipment that sends this.

bkp7, 2018-07-27:
This has already been dealt with and will be in the next release. An additional `displayScale` property has been added to meta. It will look like:
```json
"environment": {
  "depth": {
    "belowTransducer": {
      "value": 76.1,
      "timestamp": "...",
      "$source": "...",
      "meta": {
        "description": "Depth below Transducer",
        "units": "m",
        "displayScale": {
          "lower": 0,
          "upper": 100
        }
      }
    }
  }
}
```

tkurki, 2018-07-29:
I don't think these are the same? If the sounder can technically detect seabed at max 200 meters I have no use for anything beyond 30 meters and would like my display scale to be 0 to 20 m.

The full model stores values per path under context path and metadata about sources under `/sources`. This value would fall logically under source metadata more than it is associated with a value received from a transducer. We don't have yet defined a mechanism how this type of information would be output by a producer (parser) and end up under `sources`. In this respect this is similar to metadata about NMEA2000 equipment metadata, such a manufacturer info and software versions etc.