SignalK/specification #519: missing: Drift

Dirk--, 2018-10-30:
- Drift (drifting speed relative to ground) navigation.drift
- Set (drifting direction relative to true north) navigation.setTrue
- Set (drifting direction relative to magnetic north) navigation.setMagnetic

Dirk--, 2018-10-30:
DFT and SET is a movement of the vessel ergo in navigation context.
might be caused by environment.current, environment.wind, environment.gravity, ...

environment.current.drift = speed of current relative to ground
environment.current.setTrue = direction of current relative to true north

JonRowe, 2018-10-30:
In navigational terms, drift and set refer to current/tide, not wind or anything else.

joabakk, 2018-10-30:
It is sufficient to have set and drift under environment. Your boat moves in the water, and the water moves under you. That is a result of the environment. The current use is to take the current as derived from speeds and course in e.g. NMEA0183 VDR into environment.current.

Dirk--, 2018-10-31:
got it: https://en.wikipedia.org/wiki/Set_and_drift
Excuse my ignorance.

closed 2018-10-31 by Dirk--