SignalK/specification #564: Joules versus AmpHours

FransVeldman, 2020-02-26:
Hi,

When trying to output the data of my BMS I stumbled over the specs which want to have all capacity related data expressed in Joules rather than AmpHours. Apparently, someone went to great lengths to get everything into SI units, but made a serious mistake.

The problem is: Since an Amp Hour (Ah) is a measure of charge (Coulombs) whereas a Joule is a measure of energy, you can't convert mAh to Joules without knowing the voltage at which the charge was transferred. Similary, you can't convert "Joules remaining" to "Ah's remaining" without knowing the voltage at which the transfer is going to take place.

Battery manufacturers specify their battery capacity in AmpHours (at a give discharge rate). Battery Management Systems track the Amps going in and out of the battery and use it to calculate the State Of Charge. Voltage is entirely out of the equation here and would deeply confuse matters.

Let me clarify why this is creating a problem: Suppose we are discharging a 100Ah battery with 10 Amps. The battery is fully charged and starts at a voltage of 13.7 Volts. During the discharge, after one hour the voltage has dropped to 12.5V. We have used 10 AmpHours. During the next hour, the voltage drops a bit more, to 12.2V. Again we have used 10 AmpHours. Now, if we convert both hours to Joules, we will see a difference between the amount of Joules used during these two hours because the voltages during the discharge were different, yet the amount of AmpHours used is in both situations the same.

It gets even more confusing when we want to indicate how much capacity is left in the battery. It is easy in AmpHours, because we used two hours of 10Ah, so there is 80Ah left. But even if we went through the trouble to track the amount of Joules used (which was dependent on a varying Voltage), we are now stuck with a capacity expressed in Joules which we can not convert back into Ah's without knowing at forehand at which Voltage the transfer is going to take place. You see, voltage varies with temperature so when the temperature of the battery changes, the amount of Joules left changes as well, whereas the amount of Ah's left remains the same.

It is not a small error we are talking about. Battery voltage varies quite a bit, in a working installation it can easily vary between 11.5 V to 14.5 Volts. If we convert Ah's to Joules back and forth we are talking about an error easily exceeding 20%!

Technically, the Ah's is what we want to track. This is the unit specified by the battery manufacturer, and this is the unit users understand. This is also the only unit Battery Management Systems can use to track the State Of Charge. Battery voltage varies between charging and discharging, yet you can expect that when taken out 10Ah's you also need to put 10Ah back into the battery. The voltage (and hence the amount of Joules) is variable, not just due to the charge left in the battery, but also due to temperature, load, internal resistance, but although the voltage varies, it has no effect on the amount of AmpHours. The user wants to know how many amphours are left in the battery, and he doesn't expect this value to change when the temperature of the battery (and hence the voltage and thus Joules) changes.

So, I would propose to use Ah (or Coulomb) as a unit, not Joules, as there is no linear relationship between Ah's and Joules, and in battery terms Ah is the unit we are interested in, and we don't want to see this unit polluted by temperature differences and internal resistance etc.

The keys/paths affected are:
/capacity/nominal
/capacity/actual
/capacity/remaining
/capacity/dischargeLimit

tkurki, 2020-12-25:
What did you end up doing? Sorry about letting the issue lapse, I think your arguments make perfect sense.

I also just added #603 related to BMSs.

Techstyleuk, 2023-01-11:
Just putting together a battery monitor and saw this same issue.  NMEA2000 spec calls out capacity in Coulumbs:

SetN2kPGN127513(N2kMsg,BatInstance,BatType,SupportsEqual,BatNominalVoltage,BatChemistry,BatCapacity,
BatTemperatureCoefficient,PeukertExponent,ChargeEfficiencyFactor);

**********************************************************
 * \param N2kMsg       Reference to a N2kMsg Object, 
 *                     Output: NMEA2000 message ready to be send.
 * \param BatInstance          BatteryInstance
 * \param BatType              Type of battery. See \ref tN2kBatType
 * \param SupportsEqual        Supports equalization. See \ref tN2kBatEqSupport
 * \param BatNominalVoltage    Battery nominal voltage. See \ref tN2kBatNomVolt
 * \param BatChemistry         Battery Chemistry See \ref tN2kBatChem
 * \param BatCapacity          Battery capacity in Coulombs. Use AhToCoulombs,
 *                             if you have your value in Ah.
 * \param BatTemperatureCoefficient Battery temperature coefficient in %
 * \param PeukertExponent      Peukert Exponent
 * \param ChargeEfficiencyFactor      Charge efficiency factor
*********************************************************

whereas the SignalK Spec. calls out Joules:

/vessels/<RegExp>/electrical/batteries/<RegExp>/capacity/nominal
Units: J (Joule)

Description: The capacity of battery as specified by the manufacturer

Should these be common?

MikeP-NZ, 2023-08-27:
"Description: The capacity of battery as specified by the manufacturer"
Manufacturers nearly always specify battery capacity in Ah.  They often will state a kWh value, but it is usually the Ah capacity x the nominal voltage.  I've only ever come across laptop batteries that were only marked with a capacity in Wh.  Part of the reason is that Coulombic efficiency (Ah)  is less dependent on discharge rate than energy efficiency (Wh).

keesverruijt, 2026-10-03:
Bringing this back up, because the ecosystem has settled on coulomb while the schema still says joule.

What exists today:

- **NMEA 2000** gives battery capacity as a charge: PGN 127513 *Battery Configuration Status*, field *Battery Capacity*, in Ah. canboatjs 4 decodes it in SI, coulomb, and n2k-signalk 5 publishes that on `electrical.batteries.<n>.capacity.nominal`.
- **signalk-server** files `electrical.batteries.*.capacity.nominal` and `capacity.remaining` under the *charge* category, `siUnit: "C"`, in `unitpreferences/default-categories.json`. So the server's own unit conversion already disagrees with the schema: a value in J would be shown as amp-hours.
- **signalk-venus-plugin** converts every Victron amp-hour value to coulomb (`consumedCharge`, `dischargeSinceFull`, `lifetimeDischarge`).
- The schema already uses C for `capacity.dischargeSinceFull`, next to `nominal`/`remaining` in J, so one cannot be taken from the other without a voltage.

I found nobody publishing joules on these paths.

Proposal, as in the original post: change `units` from `J` to `C` for

- `electrical.batteries.<id>.capacity.nominal`
- `electrical.batteries.<id>.capacity.actual`
- `electrical.batteries.<id>.capacity.remaining`
- `electrical.batteries.<id>.capacity.dischargeLimit`

Battery *energy* (Wh/J), where a device really reports it, would then need paths of its own rather than sharing these.

Happy to open the PR against `schemas/groups/electrical.json` if this is agreed.