SignalK/specification #578: Additional battery fields

fabdrol, 2020-07-15:
Recently I needed to represent additional battery information from PGN 127513 (Battery Configuration Status). Some of these fields are not yet present. This is how the canboat output looks:

```javascript
const ex = {
   "pgn":127513,
   "description":"Battery Configuration Status",
   "fields":{
      "Instance":16,
      "Battery Type":"Gel", // electrical.batteries.16.chemistry
      "Supports Equalization":0, // NOT PRESENT IN SK
      "Nominal Voltage":"12", // NOT PRESENT IN SK
      "Capacity":0, // electrical.batteries.16.capacity.nominal
      "Temperature Coefficient":19, // NOT PRESENT IN SK
      "Peukert Exponent":0.002, // NOT PRESENT IN SK
      "Charge Efficiency Factor":0, // NOT PRESENT IN SK
   }
}
```

I would like to propose the following fields. If nobody objects, I'll do PRs for `n2k-signalk` and `specification`

- electrical.batteries.<inst>.voltage.nominal
- electrical.batteries.<inst>.temperature.coefficient
- electrical.batteries.<inst>.peukertExponent
- electrical.batteries.<inst>.chargeEfficiency
- electrical.batteries.<inst>.supportsEqualisation

Thoughts?

natecostello, 2020-12-22:
Has there been any consideration of adding data for individual cells (voltage and temperature)?  This is important for DIY battery systems.  Several BMS provide this data over CAN and it would be great to be able to capture it with Signal K.

tkurki, 2020-12-25:
@natecostello I just created #603 for that.

tkurki, 2020-12-25:
@fabdrol I have completely missed this issue, makes sense to add that.

What is the temperature coefficient? I would put that as `temperatureCoefficient`, as it is not really a "temperature".

GHBLoos, 2021-04-01:
LifePo4 batteries do not have to be equalised, but some do have balancing. 
`electrical.batteries.<inst>.supportsBalancing` = `true`, `false`, `auto` ?

Just having the support of equalisation or balancing may not be enough information. Maybe something must be said about the status: 
`isBalancing`
`isEqualizing` 
`Balancing` = `on`,` off`
`Equalizing` = `on`,` off`