SignalK/specification #593: Support for shaft rpm and remote control throttle position

phil28aus, 2020-11-12:
Working with some voyage data recorders and a couple of extra paths I think could be added that would be useful.

  /vessels/<RegExp>/propulsion/<RegExp>/shaft/rpm

Units: Hz (Hertz) 
Description: Shaft revolutions (x60 for RPM)

It would be useful to have a positive and negative value for shaft direction (CW or CCW)

***********

/vessels/<RegExp>/propulsion/<RegExp>/remoteControl/throttlePosition

Units: Ratio -1 to 1 
Description:  Throttle Position, -1<=ratio<=1, 1 is 100%

************

Shaft rpm is helpful for determining whether or not the gear box has engaged. An example is a master is coming into a wharf and goes to select astern, the throttlePosition shows that the lever has been moved to the astern position but shaft rpm is still showing a positive (or 0) value and astern gear has not been engaged.

What are everyone's thoughts

tkurki, 2020-11-12:
Just a quick comment: existing similar paths are `revolutions` - `rpm` in Hz is a bit of a wtf...

phil28aus, 2020-11-12:
That was a copy and paste mistake. I agree and have edited the original post.

hwater, 2021-04-06:
But don´t forget this, even if it is called RPM:
It would be useful to have a positive and negative value for shaft direction (CW or CCW)