SignalK/specification #595: Problem with delta PUT or delta PUT documentation or me?

preeve9534, 2020-11-14:
If I copy the example code from the documentation and tweak it a little I can make this test frame:
```
function myActionHandler(context, path, value, callback) {
   console.log(">>>>>>>>>>>>>>>>>> action handler executing");
   return { state: 'COMPLETED', statusCode: 200 };
}

plugin.start = function(options) {
  var path = "electrical.switches.bank.usb0.1.state";

  console.log(">>>>>>>>>>>>>>>> registering put handler");
  app.registerPutHandler('vessels.self', path, myActionHandler, 'somesource.1');

  var putdelta = {
     "context": "vessels.self",
     "correlationId": uuidv4(),
     "put": {
        "source": plugin.id,
        "path": path,
        "value": 1
     }
  };

  console.log(">>>>>>>>>>>>>>>> issuing put delta");
  app.handleMessage(plugin.id, putdelta);
}
```
And I would expect to see three console log messages, the last message being emitted by myActionHandler.

In fact the action handler seems never to be called and I just see the two messages from plugin.start. No error messages in the log.

I'm quite prepared to believe that I'm missing something or misunderstanding something. But what?

tkurki, 2020-11-14:
put deltas are not processed in `app.handleMessage`, there's an (sadly undocumented) `putPath(path, value, callback, source)` to do puts from plugins. 

But....is this just testing? Why would you put from a plugin to itself?

preeve9534, 2020-11-14:
Just testing. Documentation around this is rather bleak and I have not been
able to get anything working, so back to basics...

preeve9534, 2020-11-14:
Sorry to get back on this again, but I now have the test frame below.

The action handler never gets called.

(I also tried app.putSelfPath with the same result)...

```
function myActionHandler(context, path, value, callback) {
    console.log(">>>>>>>>>>>>>>>>>> action handler executing");
    return { state: 'COMPLETED', statusCode: 200 };
  }

  plugin.start = function(options) {
    var path = "electrical.switches.bank.usb0.1.state";

    console.log(">>>>>>>>>>>>>>>> registering put handler");
    app.registerPutHandler('vessels.self', path, myActionHandler, 'somesource.1');

    console.log(">>>>>>>>>>>>>>>> issuing put delta");
    app.putPath(path, 1, () => console.log("Cooee"), "somesource.1");
  }
```

sbender9, 2020-11-14:
It could help to print out the result from app.putSelfPath

```
          app.putSelfPath(path, 1, res => {
            app.debug(JSON.stringify(res))
        })
```

sbender9, 2020-11-14:
and just in case there's an in with sources, leave off the `somesource.1` from your calls.

tkurki, 2020-11-16:
With `putPath` you need the complete path from the root, including the `vessels.self` context. `putPath` returns a Promise that you can use to figure out what's happening.

```
    var path = "electrical.switches.bank.usb0.1.state";

    console.log(">>>>>>>>>>>>>>>> registering put handler");
    app.registerPutHandler('vessels.self', path, myActionHandler, 'foo');

    console.log(">>>>>>>>>>>>>>>> issuing put delta");
    app.putPath(`vessels.self.${path}`, 1, () => console.log("Cooee"), 'foo').then(x => console.log(x))
```
You can also omit the source, as mostly there is only one handler for a path.

So this was a problem related to ther server's plugin API, that is not part of the specification to be exact.

closed 2020-11-16 by tkurki