SignalK/specification #603: BMS / individual cell data for batteries

tkurki, 2020-12-25:
LiFePO4 batteries are getting more common. They require more detailed management that is typically handled by a Battery Management System. BMSs can produce data on a cell as opposed to battery level and this data is relevant if you want to monitor the performance and health of a LiFePO4 battery.

tkurki, 2020-12-25:
Per cell data:
- voltage
- temperature

BMS data:
- balancing - what data is there about balancing?
- overcurrent/overvoltage/undervoltage/undertemperature protection states and settings

GHBLoos, 2021-03-24:
I am working on a SensESP sensor specific for a LifePo4 battery (4 cell). The ESP will measure temperature and voltage for each cell and controls a relay to cut of the battery when a protection limit is reached. I am not sure current is important, but maybe it is.
To measure voltage I use an ADS1115 and a second one for temperature. By using ADS1115 you only need four wires from the ESP to the battery. 
For voltage I measure from gnd to cell 1 positive, cell 2 positive, cell 3 positive and cell 4 positive, so I read voltage of about 3.3, 6.6, 9.9 and 13.2v. By substracting cell 2 from cell 1, you calculate the voltage of cell 2 and so on. Of course you need voltage dividers.
Optional a water sensor can be added to check if there is water in the battery compartment. 
Besides battery health and protection, you can think of coulomb counting, State of Charge (SOC) and so. I am not working on that functionality yet.
But I am a lousy coder, so this may take some time. Still a lot to do.

_Voltage readings in Signal K data browser:_
[image]
_Photo of the battery_
[image]

GHBLoos, 2021-03-29:
My initial thoughts:

All  LPF battery-cell have these charasteristics (please check those):
Voltage
* nominal voltage: 3.2v
* minimum discharge voltage: 2.5v
* maximum charge voltage: 3.65v
* charge voltage 3.6v
* cut-off voltage 3.9v
Environment
* Charge Temperature 0-45C @ 35%-85% humidity
* Discharge Temperature -20-60C @ 35%-85% humidity
* Discharge Cut-off Voltage 0-45C @ 35%-85% humidity
Cell temperature
(I can't find any numbers for that yet)

A LPF battery used for 12v is a package of 4 cells. 
A LPF battery used for 24v is a package of 8 cells.
That means that you have to multiply the battery-cell voltage 4 or 8 times.

I am thinking of the following keys (starting with `/vessels/<RegExp>/electrical/batteries/<RegExp>`)

### static numbers:
`cell.nominalVoltage`
`cell.minimumDischargeVoltage`
`cell.maximumChargeVoltage`
`cell.cutOffMaximumVoltage`
`cell.cutOffMinimumVoltage` ?
`cell.overCurrentProtection`
`cell.environmentChargeMinTemperature`
`cell.environmentChargeMaxTemperature`
`cell.environmentDischargeMinTemperature`
`cell.environmentDischargeMaxTemperature`
`cell.environmentMinRelativeHumidity`
`cell.environmentMaxRelativeHumidity`
`cell.capacity`

### battery specific numbers
`numberOfCells`

### calculated static numbers
`nominalVoltage` (`numberOfCells` * `cell.nominalVoltage`)
`minimumDischargeVoltage` (`numberOfCells` * `cell.dischargeVoltage`)
`maximumChargeVoltage` (`numberOfCells` * `cell.chargeVoltage`)
`cutOffMaximumVoltage` (`numberOfCells` * `cell.cutOffMaximumVoltage`)
`cutOffMinimumVoltage` (`numberOfCells` * `cell.cutOffMinimumVoltage`)
`overCurrentProtection` ( = `cell.overCurrentProtection`)

### measured numbers
`cell[n].voltage`
`cell[n].temperature` (a little bit confused here: environment temperature and cell temperature sometimes  are mixed up)
`environment.temperature`
`environment.relativeHumidity`

### calculated numbers
`capacity` (=`cell.capacity`, although it seems to depend on environmental temperature) 
`SOC` (State of Charge)
`DOD` (Depth Of Discharge)

### status
`online` (is the battery available for switching on or off)
`switch` (is the battery switched on (relay switch)) 
`protected` (is the battery protected from being used/from being switched on)

natecostello, 2021-04-04:
A scattering of initial thoughts/comments:

This is a really good start.  I've tried to provide examples of where at least one BMS, (the REC Q) overlaps/aligns with what you've proposed.

> My initial thoughts:
> 
> All LPF battery-cell have these charasteristics (please check those):
> **Voltage**
> 
> * charge voltage 3.6v
> * nominal voltage: 3.2v
> * minimum discharge voltage: 2.5v
> * maximum charge voltage: 3.65v
> * cut-off charge voltage 3.9v

REC Q BMS has similar parameters and a few extras:
* balance end voltage (BVOL) (e.g., 3.58V)
* balance start voltage (BMIN) (e.g., 3.45V)
* cell end of charge voltage (CHAR) (e.g., 3.58V)
* cell over voltage switch off (CMAX) (e.g., 3.85V)
* cell under voltage protection switch off (CMIN) (e.g., 2.8V)
* cell under voltage discharge protection (CLOW) (e.g., 2.9V)

I'm not sure of the value of reporting a static nominal voltage, but am open to it.

> 
> **Current**
> 
> * Overcurrent
> 
REC Q BMS doesn't have any explicit over current protection as far as I can tell.  This function is provided by fusing/breakers outside of the BMS control.  BMS's that don't rely on external contactors (ie, have internal FETs) probably do enforce current limits.  The REC Q does report limits that CAN connected sources and sinks are ideally supposed to observe.

> **Environment**
> 
> * Charge Temperature 0-45C @ 35%-85% humidity
> * Discharge Temperature -20-60C @ 35%-85% humidity
> * Discharge Cut-off Voltage 0-45C @ 35%-85% humidity
>   Cell temperature
>   (I can't find any numbers for that yet)

REC Q BMS parameters:
* cell over temperature switch-off (TMAX) (e.g, 55C)
* Under temperature charging disable (TMIN) (e.g, -10C)

> 
> A LPF battery used for 12v is a package of 4 cells.
> A LPF battery used for 24v is a package of 8 cells.
> That means that you have to multiply the battery-cell voltage 4 or 8 times.

Not necessarily.  My battery design is 2P8S, so 16 cells, but 24V.  The topology is needed to get from cell number to voltage or capacity.  That said, I'm not advocating for including a topology field.  As reflected in comments below, I'm not convinced we should be calculating pack level parameters from cell level parameters.

> 
> ## Keys
> I am thinking of the following keys (starting with `/vessels/<RegExp>/electrical/batteries/<RegExp>`)
> 
> ### static numbers (constants):
> `chemistry` = LiFePO4
> `cell.voltage.nominal` = 3.2
> `cell.voltage.dischargeLimit` =2.5
> `cell.voltage.chargeLimit` = 3.6
> `cell.voltage.dischargeCutOff`
> `cell.voltage.chargeCutOff`= 3.9
> `cell.current.overCurrentProtection` =
> `cell.environment.ChargeMinTemperature` = 0C
> `cell.environment.ChargeMaxTemperature` = 45C
> `cell.environment.DischargeMinTemperature` = -20C
> `cell.environment.DischargeMaxTemperature` = 60C
> `cell.environment.MinRelativeHumidity` = 35%
> `cell.environment.MaxRelativeHumidity` = 85%
> 
[... 31 lines cut ...]
> 
I'm not sure of any guidance on operation with respect to relative humidity.  I'm also not aware of any BMS's that monitor or report it.

Regarding cell temperatures, the REC Q supports up to three temperature sensors, which aren't necessarily related to specific cells (but could).  I do know that at least one other BMS (the EPS) does report a temperature for each cell.  I think however this ends up, there it must account somehow for these two different cases.

> ### calculated numbers
> `capacity.nominal` (=`cell.capacity.nominal`)
> `capacity.actual` (=depends on environmental temperature and/or cell temperature)
> `capacity.stateOfCharge`
> `capacity.depthOfDischarge`
> `cell.<#>.capacity.stateOfCharge`
> `cell.<#>.capacity.depthOfDischarge`

I'm not sure state of charge is relevant from a cell perspective.
Is depthOfDischarge simply 1-SOC?
To this list, though not necessarily on a cell basis, I would add stateOfHealth which is reported by some BMS's.

> 
> ### status
> `online` (is the battery available for switching on or off)
> `switch` (is the battery switched on (relay switch))
> `protected` (is the battery protected from being used/from being switched on)
> `unbalanced` (cells are unbalanced)

I think more relevant than `unbalanced` is whether the BMS is `balancing`.  That said, the BMS I am most familiar with doesn't not explicitly report when it is balancing.

GHBLoos, 2021-04-05:
That's good feedback!

- I read in some articles that unbalanced cells are determined by comparing the individual state of charge (SOC) of a cell. That would mean that every cell has a SOC.
- Some batteries don't have a balancing function build in. In fact, if you stay between certain voltage charge and discharge limits, it is very rare that your cells will get unbalanced (so a way to prevent disbalancing, is to keep yout batteries between those limits). Balancing a battery 'manually' is [very easy](https://www.youtube.com/watch?v=x5ABvbbics8). 
- yes, DOD = 1 - SOC 
- Several battery specs have humidity and temperature ranges. As I understand, if you are out of these limits, parameters as discharge voltage and charge voltage limits are going to change too. Then of course it might be interesting to know if your battery compartment is very moist or extremely cold or so.

> A few of these numbers, depending on the BMS, are not static, and also would not reflect the above calculation. The REC BMS >reports/commands semi-dynamic dynamic values for the above or similar [...]
-  You are right about that. I guess protection will always be cell based. So these numbers are more informational (you might want to show values for a battery in stead of values for a cell). But even then they should be calculated using individual cell data.
- Configuration of the battery (2P8S) should be implemented too. Fully agree on that. So cell capacity is important, but from a battery point of view, you need to know the [configuration](https://postimg.cc/qzngBr8t). 

I think a github page must be created to organize this data structure, so we can discuss the individual keys and relations between them. I am not that good in github, so I have to figure out first how to do that.

MikeP-NZ, 2023-08-27:
Great discussion. If I can contribute some suggestions...
`electical.batteries.<RegExp>`

A BMS will usually report the max and min cell voltage and temperature for a battery. These stats are important for monitoring the balance state and the performance of the BMS and the cells.
Not all BMSs  will have a temperature sensor for every cell and some will have additional temperature sensors for the cell balancing circuitry, contactors and MOSFETs.
`.cells.voltage.minimum` ; maximum of measured cell voltages
`.cells.voltage.maximum` ; minimum of measured cell voltages
`.cells.temperature.maximum` ; maximum of measured cell temperatures
`.cells.temperature.minimum` ; minimum of measured cell temperatures
`.cells.<RegExp>.voltage`
`.cells.<RegExp>.temperature`
`.<RegExp>.temperature` ; temperature of named component, eg. BMS, contactor, pre-charge resistors

`.dateLastBalance` ; Date and time that the battery was last fully charged and allowed to achieve fully balanced cells.
`.cumulativeChargeLastBalance` ; The cumulative charge (cumulative Ah) value at the last balance, or a value for accumulated Ah since last balance charge.
These are useful for triggering reminders for periodic balance charging.  Maximum lithium battery lifespan is achieved by avoiding deep discharge and also by not unnecessarily charging the battery to 100%. 
Not all BMS will track charging amp-hours and discharging amp-hours separately. Some will only track accumulated amp-hours. My experience is that they increment this value at half the rate during both charging and discharging.
`.cumulativeCharge` or `.lifetimeCharge` ; 
The BMS may report "cycles" This is not how many times the battery has been charged.  It is derived by cumulativeCharge/NominalCapacity
`.cycles`

I hope that input is helpful.

lphooge, 2024-05-14:
Note: numberOfCells can be different values than 4 or 8.
For example for electric motors 48V batteries are common, those have 16 LiFePo4 cells in series.

And in the future other cell chemistries might have even different requirements.