SignalK/specification #639: Missing unit for directionGround

jobarne, 2022-02-20:
When running signalk nodejs server and an Airmar WX220 weather station, wind data comes in as environment.self.directionGround but with no unit specified. 

Is there a way of updating the specification so that the unit is set correctly (to rad)?

jobarne, 2022-03-15:
Anyone?

tkurki, 2022-03-15:
How do you get the data into the server? Because there is no `environment.self.directionGround` (let alone environment.**self**.directionGround) defined in the current Signal K specification. 

There are 2 paths for absolute wind direction, `directionTrue` and `directionMagnetic`.

jobarne, 2022-03-15:
Thanks for your reply! 

I am sorry. It should be environment.wind.directionGround , I assume the server uses the individual js files in n2k-signalk pgns directory, There is a environment.wind.directionGround mapping in 130306.js file but I do not understand how the unit is specified...