SignalK/specification #647: Have meta Timeout = 0 equal to No Timeout

godind, 2022-12-20:
Timeout being potentially a critical meta value, we need a way to enforce 'no timeout'.

### Current State

- In principale, it is up to the data publisher to set meta timeout for data path (not done in most cases).
- In the absence of a timeout key it is up to the consumer to decide if and when data should expire.
- The only way to interject is to add/set a timeout key (in baseDelta.json for instance). This is an overiding value but, there is no way to enforce a "no timeout" rule. No rule (no timeout key) does not mean the data should not timeout. It means it's unspecified. Unspecified is important as it could lead to other logical operations eg it's not defined: warn operator to validate, etc.

### Proposition
Have Timeout key with value of 0 `timeout: 0` interpreted as 'Infinity/No timeout. Note that values `null` and `undefined` have 'special' meanings and can cause trouble.

This would means:
- In absence of the key: It's up to the consumer to take a decision
- Key present: Value expiration (in sec) is suggested
- Key present: Value of 0, not expiration is suggested (unless replaced by a new data path value, which in turn should not expire)

Not only does it make numerical sense, it's also easier the handle and validate with Form Controls and eliminates the need to delete the key to have no rule.

tkurki, 2023-01-01:
What trouble do you see `null` creating? In general it is a special value that indicates the absence of a real value and as such would be a good fit for this purpose.

godind, 2023-01-01:
`null` and `undefined` can create unforeseen interpretation side-effect for certain language, framework, libraries, etc. As an example, if you PUT `value: undefined` against a meta property, the server will deleted the key, but this is actually an unexpected side effect - did we think undefined would delete? Will null do the same thing? I think it's a risk not worth taking... but I'm no expert on the subject.

Back to the proposition at hand; `timeout: 0` is a clear and affirmative statement eg. we have a timeout and it's value is never. `timeout: null` means timeout value has no reference - not as clear a statement, regardless of the fact that the `null` keyword might be sensitive or not.

> The undefined value is a primitive value used when a variable has not been assigned a value. The null value is a primitive value that represents the null, empty, or non-existent reference

tkurki, 2023-01-02:
1. there is no special case `undefined`: firstly it can not be represented in JSON that most of current SK depends on, secondly a client needs to handle the situation that a meta property is "undefined", which in practise is that it is missing / non existent
2. `null` is already [used in Signal K](https://signalk.org/specification/1.7.0/doc/data_model.html?highlight=null#missing-or-invalid-data) and the proposed usage is imho very well in line with that, to me it would seeing a null value would be "timeout is explicitly missing, not just absent in the data, so don't apply a timeout for this path"
3. Imho zero value has actually more potential for unforeseen interpretation side-effects (example: if I just blindly apply the timeout 0 can easily mean that a value is timed out immediately). This needs documentation and taken into account in implementation in any case.

PS. I edited the title so that it describes the use case instead of prescribing a solution.

godind, 2023-01-02:
Make sense. Thanks!

godind, 2023-12-13:
Not pertinent anymore.

closed 2023-12-13 by godind