SignalK/specification #655: Add data attribute to Alarm object used in Notification messages

panaaj, 2023-04-24:
**Current state:**
Currently the alarm object used in notification messages lacks the ability to convey values (other than in the `message` text).
Consider the standard alarm MOB (man overboard), one would expect that the position at which the alarm was raised is included in the alarm object. instead the position can only be added as text in the `message` which is likely to vary from implementation to implementation and making it more difficult to parse.
```typescript
{
    "method": ["visual", "sound"],    
    "state": "emergency",    // enum string value
    "message": "Man Overboard!  latitude: 60.2345, longitude: 14.1234"   // position included in message text
}
```

**Proposal:**
- Add an attribute to the `Alarm` object used in notifications to hold data relevant to an alarm
- Define minimum required attributes for the standard alarms.

_STANDARD ALARMS:_
- 'mob'
- 'fire'
- 'sinking'
- 'flooding'
- 'collision'
- 'grounding'
- 'listing'
- 'adrift'
- 'piracy'
- 'abandon'

_Example: vessels.self.notifications.mob_
```typescript
{
    "method": ["visual", "sound"],    
    "state": "emergency",   
    "message": "Man Overboard!",
    "data": {      // optional object to hold addiditonal alarm information
        "position": {"latitude": 60.2345, "longitude": 14.1234}
    }
}
```

bergie, 2025-09-15:
This is a very good idea. Now in signalk-meshtastic we try to find the corresponding "vessel" from Signal K data model to find the position:
https://github.com/meri-imperiumi/signalk-meshtastic/commit/609f55312de720a55aaaa2e39e099e76c9e79284

We could maybe experiment with this in https://github.com/meri-imperiumi/signalk-mob-notifier which raises `notifications.mob` when it sees beacons

VladimirKalachikhin, 2025-09-16:
There can be more than one MOB, so the "data" must be a geoJSON object.

VladimirKalachikhin, 2025-09-16:
Does only one MOB drop from each mmsi?

bergie, 2025-09-16:
Each MOB beacon should have its own unique MMSI

bergie, 2025-09-16:
Non-beacon MOBs are out of scope for signalk-mob-notifier. Assumedly you'd want to raise those in some app like Freeboard and record the position there.

Some fancy navigation systems try to estimate the drifting of the mob based on wind and current conditions.

panaaj, 2025-09-17:
Freeboard-SK does add a position.
It also ensures the notification has an id so it can be cleared, silenced etc.

bergie, 2025-09-18:
@panaaj is the format compatible to the one proposed here?

Here's [how it looks like](https://github.com/meri-imperiumi/signalk-mob-notifier/blob/4d347671dbbca832a115522f92a38f4887adeffe/plugin/index.js#L134) with the mob-notifier plugin:

```
path: `notifications.mob.${mmsi}`,
value: {
  message,
  data: {
    mmsi,
    position: mobPosition,
  },
  state: 'emergency',
  method: ['visual', 'sound'],
},
```

I'm happy to make modifications to align with Freeboard-SK if needed, but then we should document the format in spec.

VladimirKalachikhin, 2025-09-19:
But I won't do it.
The MOB delta that is sent by the [GaladrielMap SK ed.](https://www.npmjs.com/package/galadrielmap_sk) to the SignalK server:
```
{
  "context": "vessels.self",
  "updates": [
    {
      "values": [
        {
          "path": "notifications.mob",
          "value": {
            "method": [
              "visual",
              "sound"
            ],
            "state": "emergency",
            "message": "A man overboard!",
            "source": "-3c416858-a1a8-4a39-aa8f-37cf81f8c168",
            "position": {
              "type": "FeatureCollection",
              "features": [
                {
                  "type": "Feature",
                  "properties": {},
                  "geometry": {
                    "type": "LineString",
                    "coordinates": []
                  }
                },
                {
                  "type": "Feature",
                  "properties": {
                    "mmsi": "123456789",
                    "safety_related_text": "",
                    "current": true
                  },
                  "geometry": {
                    "type": "Point",
                    "coordinates": [
                      28.250633,
                      61.079483
                    ]
                  }
                }
              ],
              "properties": {
                "timestamp": 1758298923
              }
            }
          }
        }
      ],
      "source": {
        "label": "galadrielmap_sk"
      },
      "timestamp": "2025-09-19T16:22:03.000Z"
    }
  ]
}
```
All this data is necessary.

The GaladrielMap SK understands the data in the format Freeboard SK, but always overwrites it with its own.

bergie, 2025-09-19:
I have no idea why you'd need whole GeoJSON there, especially with multiple features like an empty LineString

bergie, 2025-09-20:
If I understand correctly, those additional things you're showing there (and including in the notification data model) are related to rendering features of your chartplotter app. I think the app should calculate those on the fly, and the notification only include the actual position of the MOB.